enjoy-digital/litex #3: litex has circular dependencies

mithro, 2016-05-24:
* litex depends on litedram
 * litedram depends on litex

See below;
```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "litex/test_sim/conda/lib/python3.5/site-packages/litex-0.1+25.g868bc5d-py3.5.egg/litex/boards/targets/sim.py", line 10, in <module>
    from litex.soc.integration.soc_sdram import *
  File "litex/test_sim/conda/lib/python3.5/site-packages/litex-0.1+25.g868bc5d-py3.5.egg/litex/soc/integration/__init__.py", line 2, in <module>
    from litex.soc.integration.soc_sdram import SoCSDRAM
  File "litex/test_sim/conda/lib/python3.5/site-packages/litex-0.1+25.g868bc5d-py3.5.egg/litex/soc/integration/soc_sdram.py", line 8, in <module>
    from litedram.frontend import crossbar
ImportError: No module named 'litedram'
```

mithro, 2016-05-24:
Also looks like the same happens with liteeth....

enjoy-digital, 2016-05-24:
There is effectively circular dependency for targets, but that's expected.
I don't understand how you are triggering this issue?
Here by installing litex and then all others cores everything is fine.

enjoy-digital, 2017-01-19:
Can we close this? What are you suggesting otherwise?

closed 2017-01-25 by enjoy-digital