enjoy-digital/litex #23: Option to change toolchain path?

GuzTech, 2017-03-21:
I'm trying to build Arty-SoC but I'm getting an error that the Xilinx tools were not found in /opt/Xilinx/Vivado which makes sense because I installed it at $HOME/Software/Xilinx/Vivado. Other than changing the hardcoded path, is there a way to point to a different toolchain path on the command line as an argument?

mithro, 2017-03-21:
Try `-Ob toolchain_path $XILINX_DIR/opt/Xilinx/` ?

GuzTech, 2017-03-21:
@mithro Where should I put those arguments?

I tried `make arty_minisoc -Ob toolchain_path /home/guztech/Software/Xilinx/` in the arty-soc repo, and `sudo python3.5 setup.py develop -Ob toolchain_path /home/guztech/Software/Xilinx/` in the litex repo, but none of them worked.

mithro, 2017-03-22:
@GuzTech Sorry for the red herring, it doesn't look like arty-soc has support for these options out of the box.

@GuzTech You might want to take a look at our [`make.py`](https://github.com/timvideos/HDMI2USB-litex-firmware/blob/master/make.py) which allows you to use those options like that.

enjoy-digital, 2017-04-26:
@GuzTech: I'm closing this since this is related to https://github.com/enjoy-digital/arty-soc repo, please re-open one there if you still need it.

closed 2017-04-26 by enjoy-digital