enjoy-digital/litex #33: simulation shouldn't run as root user

felixheld, 2017-10-16:
https://github.com/enjoy-digital/litex/blob/db6c88bbef0bdd64890e38aacf069f1f29d675ee/litex/build/sim/verilator.py#L129

The simulation still works with the sudo being removed. I'd avoid having code running as root when it's not necessary.

mithro, 2017-10-16:
I generally agree, however root is needed for things like network access or block direct access.

Definately welcome to suggestions & fixes!

felixheld, 2017-10-16:
For raw socket support you can use CAP_NET_RAW. Much safer than running some binary as root.

q3k, 2018-01-22:
Network access could also be done via a user networking stack like the one in QEmu, as long as having TCP/UDP support only is enough (as other protocols common protocols are privileges).

Ripping SLIRP out of the QEmu codebase is quite a challenge right now, though.

mithro, 2018-01-23:
We have user mode network in qemu this way -> https://github.com/timvideos/litex-buildenv/blob/master/scripts/build-qemu.sh#L188-L204

enjoy-digital, 2018-08-22:
With https://github.com/enjoy-digital/litex/commit/09776b77e649ebc7138baa9a355fb42931fe915a simulation run as root only when really needed (ie when ethernet module is present). This seems a reasonable trade-off.

closed 2018-08-22 by enjoy-digital