enjoy-digital/litex #62: Installation error on CentOS 6.9 with Python 3.4

ShawnLess, 2018-01-25:
When I typed following command to install Litex:
```bash
sudo python3 setup.py install
```
Some errors propmted and It seems some packages were still using python 2 syntax ?
```bash
 File "/usr/lib/python3.4/site-packages/litex-0.1-py3.4.egg/litex/build/sim/core/modules/ethernet/tapcfg/drivers/osx/tuntap/test/tuntap/interface_harness.py", line 231
    return filter(lambda (a, n, d): a != None,
                         ^
SyntaxError: invalid syntax

  File "/usr/lib/python3.4/site-packages/litex-0.1-py3.4.egg/litex/build/sim/core/modules/ethernet/tapcfg/drivers/osx/tuntap/test/tuntap/route.py", line 85
    def add_addr((addr_flags, payload), (addr, flag)):
                 ^
SyntaxError: invalid syntax

  File "/usr/lib/python3.4/site-packages/litex-0.1-py3.4.egg/litex/build/sim/core/modules/ethernet/tapcfg/scons-tools/crossmingw.py", line 88
    raise SCons.Errors.UserError, "A shared library should have exactly one target with the suffix: %s" % env.subst("$SHLIBSUFFIX")
...
```

enjoy-digital, 2018-01-26:
I'll look at that, it seems tapcfg submodule has python2 files in it. You can use develop instead of install for now.

enjoy-digital, 2018-02-19:
Errors reported are due to submodules that are using python 2 syntax. We can ignore them. Prefered method is to use setup.py develop.

closed 2018-02-19 by enjoy-digital