enjoy-digital/litex #73: Package installs 'test' package

blshkv, 2018-06-01:
I'm writing an ebuild for Pentoo (Gentoo) and facing an issue similar to the https://github.com/ypid/yaml4rst/issues/1:

```
/usr/bin/python3.5 -O /var/tmp/portage/sys-apps/litex-20180531/temp/tmphmwkkika.py
removing /var/tmp/portage/sys-apps/litex-20180531/temp/tmphmwkkika.py
writing byte-compilation script '/var/tmp/portage/sys-apps/litex-20180531/temp/tmptgofa80_.py'
/usr/bin/python3.5 -OO /var/tmp/portage/sys-apps/litex-20180531/temp/tmptgofa80_.py
removing /var/tmp/portage/sys-apps/litex-20180531/temp/tmptgofa80_.py
running install_egg_info
Copying litex.egg-info to /var/tmp/portage/sys-apps/litex-20180531/image/_python3.5/usr/lib64/python3.5/site-packages/litex-0.2.dev0-py3.5.egg-info
running install_scripts
Installing mkmscimg script to /var/tmp/portage/sys-apps/litex-20180531/image/_python3.5/usr/lib/python-exec/python3.5
Installing litex_term script to /var/tmp/portage/sys-apps/litex-20180531/image/_python3.5/usr/lib/python-exec/python3.5
Installing litex_server script to /var/tmp/portage/sys-apps/litex-20180531/image/_python3.5/usr/lib/python-exec/python3.5
 * ERROR: sys-apps/litex-20180531::pentoo failed (install phase):
 *   Package installs 'test' package which is forbidden and likely a bug in the build system.
 * 
 * Call stack:
 *     ebuild.sh, line  124:  Called src_install
 *   environment, line 2604:  Called distutils-r1_src_install
 *   environment, line  835:  Called _distutils-r1_run_foreach_impl 'distutils-r1_python_install'
 *   environment, line  334:  Called python_foreach_impl 'distutils-r1_run_phase' 'distutils-r1_python_install'
 *   environment, line 2151:  Called multibuild_foreach_variant '_python_multibuild_wrapper' 'distutils-r1_run_phase' 'distutils-r1_python_install'
 *   environment, line 1526:  Called _multibuild_run '_python_multibuild_wrapper' 'distutils-r1_run_phase' 'distutils-r1_python_install'
 *   environment, line 1524:  Called _python_multibuild_wrapper 'distutils-r1_run_phase' 'distutils-r1_python_install'
 *   environment, line  546:  Called distutils-r1_run_phase 'distutils-r1_python_install'
 *   environment, line  803:  Called distutils-r1_python_install
 *   environment, line  725:  Called die
 * The specific snippet of code:
 *               die "Package installs '${p}' package which is forbidden and likely a bug in the build system.";
 * 
 * If you need support, post the output of `emerge --info '=sys-apps/litex-20180531::pentoo'`,
 * the complete build log and the output of `emerge -pqv '=sys-apps/litex-20180531::pentoo'`.
 * The complete build log is located at '/var/tmp/portage/sys-apps/litex-20180531/temp/build.log'.
 * The ebuild environment file is located at '/var/tmp/portage/sys-apps/litex-20180531/temp/environment'.
 * Working directory: '/var/tmp/portage/sys-apps/litex-20180531/work/litex-c534250c43791612da2a03c1c93f992115d11f8c'
 * S: '/var/tmp/portage/sys-apps/litex-20180531/work/litex-c534250c43791612da2a03c1c93f992115d11f8c'
```

Could you help to fix it please?

enjoy-digital, 2018-06-01:
test is not suppored to be installed, can you try removing the __init__.py file and retry the installation? If that works i'll probably delete it.

blshkv, 2018-06-01:
Please go ahead and delete ```__init__.py```. It fixed the issue.
Thank you

enjoy-digital, 2018-07-18:
This should be fixed by https://github.com/enjoy-digital/litex/commit/b19844d1964432b3e8b768e3698edab84aa3a2f8.

closed 2018-07-18 by enjoy-digital