enjoy-digital/litex #82: Accessing unmapped Wishbone address freezes system

xobs, 2018-08-02:
The Wishbone bus is sparsely populated, and contains many unmapped addresses.  Attempting to access one of these addresses locks up the system.

There are at least two issues here:

1. There is nothing listening on the "null" `cyc` line, so there is no default listener
2. The current Wishbone architecture assumes transactions are always successful, and will wait forever if a slave pulls `err`.

For (1), it should be a simple matter of coming up with a signal that feeds into the `err` (or `ack`) signal on Wishbone that is `(~(slave[0].cyc | slave[1].cyc | ... | slave[n].cyc) & slave.stb)`, such that a response signal is generated immediately upon access to an unmapped address.

For (2), the state machine will need to be modified somewhat to return to the `IDLE` state if `err` is hit, or however Wishbone is supposed to handle error conditions.

Because (1) is much easier to do than (2), it may be easier to have invalid accesses return bogus data than to have invalid accesses return an error condition.  Either way, it would be nice to not have Wishbone lock up when accessing an invalid address.

enjoy-digital, 2018-08-02:
Thanks, yes we should improve that. I'll look at that when in a few days.

xobs, 2018-08-03:
I have a simple, untested attempt at https://github.com/xobs/litex/commit/033be858ebe361dabb61a1fb4249d813d2ce8106

enjoy-digital, 2018-08-03:
I'll do the review/test/integration next week. That's a good idea to have a counter error, we should then expose it. I'll maybe add a specific module for Wishbone/CSR  testing that will report access errors and also add some scratchpad registers to test Wishbone and CSR accesses. (would be useful for software developers to be sure that wishbone and registers are accessed correctly by the software).

enjoy-digital, 2018-08-06:
I added a Timeout module to cover that. If an access is not acked for a certain duration (configurable), it's acked automatically by the Timeout module and an error is generated/counted.

The error count is reported by the controller of the SoC (just added), that also enables reseting the CPU and provides a scratch register for CSR access testing when developing software/drivers.

closed 2018-08-06 by enjoy-digital