enjoy-digital/litex #85: Possible problem with DCACHE when using VexRiscv

pgielda, 2018-08-12:
I am using VexRiscv version of LiteX on Arty board.
I've discovered that any software running from DDR (from 0x40000000) via uploading a binary using flterm will after executing a few instructions crash in an exception with mbadaddr pointing to the place where it got stuck (usually 0x40000120) with mcause == 2 (unknown instruction).
This can be fixed (at least seems so) by cleaning DCACHE three times (no less, otherwise it will crash but somewhere further in the way):

diff --git a/litex/soc/software/bios/boot.c b/litex/soc/software/bios/boot.c
index 9e5a4dc..bf6c698 100644
--- a/litex/soc/software/bios/boot.c
+++ b/litex/soc/software/bios/boot.c
@@ -26,6 +26,9 @@ static void __attribute__((noreturn)) boot(unsigned int r1, unsigned int r2, uns
 	irq_setmask(0);
 	irq_setie(0);
 	flush_cpu_icache();
+	flush_cpu_dcache();
+	flush_cpu_dcache();
+	flush_cpu_dcache();
 	boot_helper(r1, r2, r3, addr);
 	while(1);
 }

Since this is a generic part of LiteX I assume this has to be a bug in VexRiscv cache handling (or I am completely misunderstanding something).

The way I generate the core is:
python3 litex/boards/targets/arty.py --cpu-type vexriscv --csr-csv `pwd`/csr.csv

enjoy-digital, 2018-08-12:
Thanks, i'll try to reproduce. Do you have a minimal firmware to load with flterm that trigger the issue? (On others project we are running software from DDR with VexRiscv, so that would help understand what is going on).

pgielda, 2018-08-12:
Yes, I know this looks suspicious.

I've prepared a minimal test for that:
https://github.com/pgielda/litex-baremetal-test

This is a very minimal *standalone* baremetal program. The only trick there is that I "steal" `flush_cpu_dcache` so you will need to update the address in linker.ld with a correct one.

Just tested this and it only works if the `flush_cpu_dcache` is called 3 times at the beginning. Otherwise the firmware reboots the board (which means that mcause was != 0).

If you do not want to update linker.ld you can also remove `flush_cpu_dcache` calls from my code, test and prove that the code reboots itself, then add those to boot.c in the BIOS and see that the code stops rebooting.

At least this happens on my setup with ARTY.

enjoy-digital, 2018-08-16:
It seems flush dcache and l2_cache make it work, can you test with upstream LiteX?

pgielda, 2018-08-17:
Thanks, this change looks promising. I will test this over the weekend.

enjoy-digital, 2018-08-17:
OK thanks.

pgielda, 2018-08-18:
I can confirm that his fixes the problem.

closed 2018-08-18 by pgielda