enjoy-digital/litex #102: LiteX and Migen relationship and compatibility is unclear

abraxa, 2018-09-20:
Hi, I'm trying to wrap my head around the LiteX/Migen world because I want to get LiteScope running on my Lattice HX8K board. However, this is unfortunately turning out to be a rather frustrating experience.

On https://github.com/enjoy-digital/litex, it says "All Migen's platforms can also be used in LiteX" - which makes sense as https://www.bunniestudios.com/blog/?p=5018 mentions that "LiteX is a soft-fork of Migen/MiSoC". However, it is unclear to me why LiteX was created (instead of improving on Migen) and what the differences are. I would appreciate if those points were clarified in readme.md.

Anyway, I'd like to use https://github.com/m-labs/migen/blob/master/migen/build/platforms/ice40_hx8k_b_evn.py but when I specify it as a platform to LiteX, it complains that it doesn't know about it - which makes sense because litex and migen use different python name spaces. So how, then, can Migen platforms be used in LiteX? As a newbie, I'm not sure how this can be achieved and would appreciate a more precise expression such as "All of Migen's platforms can also be used in LiteX with minor modifications" - or an explanation of how to use the platforms that only Migen currently supports.

Thanks for your hard work!

enjoy-digital, 2018-09-21:
Hi,

i added some clarification here: https://github.com/enjoy-digital/litex/commit/01ae7d4235d141f4965ccaff02d61c76c5ff662a

Initially, LiteX was created as a soft fork of Migen/MiSoC as a way for us to keep things working since a number of features we needed were arbitrary removed from Migen/MiSoC and i needed some time to figure out the best way to handle that. Things are more clear now and LiteX is based on Migen and is an alternative to MiSoC with additionnal tools/features and is compatible with the LiteX cores ecosystem.

LiteScope is only based on Migen. (LiteX is used to create full designs/SoCs) and i just added an option to use platforms from Migen if platform is not defined in LiteX.

In Litescope/examples, to build the example with your migen platform, you can now run:
`python3 -t simple -p ice40_hx8k_b_evn build-bitstream`

I hope it's clear now for you. Please ask if not.

abraxa, 2018-09-21:
Thank you, it works now and makes it a lot clearer! :)
"LiteX is based on Migen and is an alternative to MiSoC with additionnal tools/features" - this sentence is essential to the understanding, though, and I would suggest you add it to the readme as well.

I'll continue working on a basic LiteScope setup then. My goal is to create a "basic" logic analyzer using only LiteScope and a Wishbone<->UART bridge (maybe https://zipcpu.com/blog/2017/06/05/wb-bridge-overview.html) so that it can be used on all barebones FPGA boards and doesn't require LiteEth.

enjoy-digital, 2018-09-21:
What you are trying to do is already what the example is doing :):
https://github.com/enjoy-digital/litescope/blob/master/examples/targets/simple.py#L30

enjoy-digital, 2018-09-21:
BTW, if you just want to create a core to use directly in your verilog design with also the wishbone<-->uart bridge integrated, you can have a look at that: https://github.com/enjoy-digital/netv2-soc/blob/master/gateware/litescope/build.py

abraxa, 2018-09-21:
I think it's pretty obvious I didn't get very far yet, so thanks for the hint ;)
Indeed the example appears to do exactly what I need. I'll try to build it for my HX8K board, flash it and then try to interface it to the sigrok driver which currently only supports the LiteEth interface as far as I can see. Ideally, I'd like the driver to support both ethernet and UART.

enjoy-digital, 2018-10-02:
I think it's more clear now, let's discuss project specific issues in Litescope repo if any issues.

closed 2018-10-02 by enjoy-digital