enjoy-digital/litex #103: Integrate more RISC-V cores and compare resources/max freq of all cores.

enjoy-digital, 2018-09-21:
PicoRV32, VexRiscv are already supported, good candidates SRC1 (Syntacore) / RISCY (Pulp).

mithro, 2018-09-21:
RISCY is a SystemVerilog core, so might not be a great option.

SRC1 seems like a good idea.

Other options might be;
 * https://github.com/rgwan/bapi-rv32i
 * https://github.com/AnlogicInfo/ridecore
 * Hummingbird?

enjoy-digital, 2018-09-21:
Thanks for the suggestions. While doing this work, we could also think about how to simplify new RISC-V core integration.

mithro, 2018-09-21:
I think we should have some way of providing a CPU using a plugin type system?

I also interested in supporting the CPU variant better. See my test examples at https://github.com/enjoy-digital/litex/compare/master...mithro:cpu-config

@cr1901 was also interested in potential SH2 support.

enjoy-digital, 2019-04-28:
For FPGA projects, VexRiscv allow us to create all the variants we need. We also have Minerva and PicoRV32 integrated, closing since we don't have a real need for others cores for now, we'll re-open if it's the case.

closed 2019-04-28 by enjoy-digital