enjoy-digital/litex #107: SH2/J-core support

cr1901, 2018-09-22:
The free SH2 core- aka J-core- is written in VHDL. In principle, J-core/SH2 should work already with Xilinx cores- someone just needs to port the Wishbone interface, startup code, etc :).

For ice40 toolchains, there is essentially no VHDL toolchain that targets yosys. I was fortunate enough to talk to J-core's creator on IRC, and he had this to say:

* `yosys` support is something they've wanted for a while, but J-core uses many features of VHDL that don't translate to Verilog as-is.
* The `nvc` [VHDL simulator](https://github.com/nickg/nvc) is [already capable](http://lists.j-core.org/pipermail/j-core/2016-June/000184.html) of simulating J-core.
* Synthesizing VHDL is more complicated than Verilog, to the extent that you essentially need a simulator to instantiate parts of a VHDL design (this is known as VHDL elaboration).
* The best way forward is to extend `nvc` to output Verilog at a lower-level than users would normally write. `nvc`'s LLVM backend can handle VHDL elaboration that needs to occur before emitting Verilog.

Link to the full conversation: http://ix.io/1ngU

enjoy-digital, 2018-09-22:
You could already had support for official toolchains and then figure out how to support VHDL with free/open toolchains.

cr1901, 2018-09-22:
@enjoy-digital That's fine for Xilinx/Altera/Diamond boards. I don't plan on supporting Icecube/Radiant however in Migen/LiteX, so ice40 boards would be left out for now.

enjoy-digital, 2019-04-28:
Closing since no-update, please re-open if you still have a need/want to work on that.

closed 2019-04-28 by enjoy-digital