enjoy-digital/litex #115: cpu_interface: select_triple silently returns None

JohnSully, 2018-10-05:
Related to this change here:
https://github.com/enjoy-digital/litex/commit/b528a005a04b187f54d0c3783736382e2caa6ab9

Instead of a useful error message compilation fails with:
  File "core_nopll.py", line 247, in <module>
    main()
  File "core_nopll.py", line 226, in main
    vns = builder.build(build_name="litedram_core", regular_comb=False)
  File "/home/john/repos/litex/litex/soc/integration/builder.py", line 156, in build
    self._generate_includes()
  File "/home/john/repos/litex/litex/soc/integration/builder.py", line 72, in _generate_includes
    define(k, v)
  File "/home/john/repos/litex/litex/soc/integration/builder.py", line 70, in define
    variables_contents.append("{}={}\n".format(k, _makefile_escape(v)))
  File "/home/john/repos/litex/litex/soc/integration/builder.py", line 27, in _makefile_escape
    return s.replace("\\", "\\\\")
AttributeError: 'NoneType' object has no attribute 'replace'

JohnSully, 2018-10-06:
Note: you get this when it can't find the toolchain.  It should give an error message saying that like the old code did.

enjoy-digital, 2018-10-06:
Thanks, i don't see how you get this, can you provide a minimal code to reproduce?

JohnSully, 2018-10-06:
Remove your riscV compiler/tool chains from your PATH or otherwise prevent
litex from finding them. Then try to build a litex SoC with a picorv32.

JohnSully, 2018-10-06:
Also for full explicitness, the expected behavior in this case is an error message saying it can't find the toolchain.  The original code used to have a mechanism for that.

enjoy-digital, 2018-10-06:
Fixed via https://github.com/enjoy-digital/litex/commit/3b27d2ae895aa3045759a3716aee07f97597dd0e.

closed 2018-10-06 by enjoy-digital