enjoy-digital/litex #132: Allow Yosys to be used for synthesis with Vivado

enjoy-digital, 2018-12-12:
Reopening https://github.com/enjoy-digital/litex/pull/114 as an issue:

The aim here is to allow Yosys to be used for the synthesis step while Vivado remains being used for place and route. (Eventually I hope to also replace Vivado with something based on nextpnr or vpr.)
- [ ] Add a (* keep *) attribute to IDELAYCTRL,
- [ ] Check no VHDL files are in the source output.
- [ ] Add the yosys step to the build_top.sh script.
- [ ] Actually test the output bitstream.
- [ ] Compare Vivado and Yosys results for synthesis.

enjoy-digital, 2018-12-12:
Simple designs are working: led blink, uart bridge. But a design with a cpu executing code from ROM isn't working. I need to do more testing, but it could be related to block ram support in Yosys for 7-series. (simular designs are working fine on iCE40 or ECP5).

enjoy-digital, 2019-10-06:
@mithro, @daveshah1: just curious, have you been able to test Yosys synthesis + Vivado P&R recently with a LiteX SoC?

mithro, 2019-10-06:
@eddiehung has been working on this I believe...

mithro, 2019-10-06:
@mkurc-ant has also been testing at https://github.com/SymbiFlow/prjxray/tree/master/minitests/litex

daveshah1, 2019-10-06:
Yes, I had linux-on-litex-vexriscv working on the Arty here: https://github.com/daveshah1/linux-on-litex-vexriscv/tree/arty-yosys (even got it working with nextpnr removing XADC and reducing frequency)

daveshah1, 2019-10-06:
As far as I know the issue was previously blocking Yosys and Vivado with LiteX was an EDIF flavour issue, I submitted a LiteX PR which was merged and fixed this a while ago.

enjoy-digital, 2019-10-06:
@mithro, @daveshah1: thanks for the quick feedback. I'll do a test tomorrow then and will implement the few points we were listing.

mkurc-ant, 2019-10-06:
I can also confirm that Litex with VexRiscv+Ethernet+DDR works on Arty when synthesized by Yosys and P&R-ed with Vivado. There is also a fork of the main LiteX repo with modified `arty.py` that allows to select synthesis engine between Vivado and Yosys: https://github.com/antmicro/litex

The EDIF backend issue came down to specifying correct bit ordering in vectors through `write_edif -pvector bra`.

enjoy-digital, 2019-10-07:
Arty BaseSoC (DDR3, Vexriscv) with Vivado synthesis / Vivado P&R:
````

1. Slice Logic
--------------

+----------------------------+------+-------+-----------+-------+
|          Site Type         | Used | Fixed | Available | Util% |
+----------------------------+------+-------+-----------+-------+
| Slice LUTs                 | 4144 |     0 |     20800 | 19.92 |
|   LUT as Logic             | 3998 |     0 |     20800 | 19.22 |
|   LUT as Memory            |  146 |     0 |      9600 |  1.52 |
|     LUT as Distributed RAM |  144 |     0 |           |       |
|     LUT as Shift Register  |    2 |     0 |           |       |
| Slice Registers            | 3183 |     0 |     41600 |  7.65 |
|   Register as Flip Flop    | 3183 |     0 |     41600 |  7.65 |
|   Register as Latch        |    0 |     0 |     41600 |  0.00 |
| F7 Muxes                   |   34 |     0 |     16300 |  0.21 |
| F8 Muxes                   |    8 |     0 |      8150 |  0.10 |
+----------------------------+------+-------+-----------+-------+

1.1 Summary of Registers by Type
--------------------------------

+-------+--------------+-------------+--------------+
| Total | Clock Enable | Synchronous | Asynchronous |
+-------+--------------+-------------+--------------+
| 0     |            _ |           - |            - |
| 0     |            _ |           - |          Set |
| 0     |            _ |           - |        Reset |
| 0     |            _ |         Set |            - |
| 0     |            _ |       Reset |            - |
| 0     |          Yes |           - |            - |
| 4     |          Yes |           - |          Set |
| 0     |          Yes |           - |        Reset |
| 106   |          Yes |         Set |            - |
| 3073  |          Yes |       Reset |            - |
+-------+--------------+-------------+--------------+

2. Slice Logic Distribution
---------------------------

+-------------------------------------------+------+-------+-----------+-------+
|                 Site Type                 | Used | Fixed | Available | Util% |
+-------------------------------------------+------+-------+-----------+-------+
| Slice                                     | 1278 |     0 |      8150 | 15.68 |
|   SLICEL                                  |  889 |     0 |           |       |
|   SLICEM                                  |  389 |     0 |           |       |
| LUT as Logic                              | 3998 |     0 |     20800 | 19.22 |
|   using O5 output only                    |    1 |       |           |       |
|   using O6 output only                    | 3428 |       |           |       |
|   using O5 and O6                         |  569 |       |           |       |
| LUT as Memory                             |  146 |     0 |      9600 |  1.52 |
|   LUT as Distributed RAM                  |  144 |     0 |           |       |
|     using O5 output only                  |    0 |       |           |       |
|     using O6 output only                  |    0 |       |           |       |
|     using O5 and O6                       |  144 |       |           |       |
|   LUT as Shift Register                   |    2 |     0 |           |       |
|     using O5 output only                  |    2 |       |           |       |
|     using O6 output only                  |    0 |       |           |       |
|     using O5 and O6                       |    0 |       |           |       |
| LUT Flip Flop Pairs                       | 1296 |     0 |     20800 |  6.23 |
|   fully used LUT-FF pairs                 |  188 |       |           |       |
|   LUT-FF pairs with one unused LUT output | 1004 |       |           |       |
|   LUT-FF pairs with one unused Flip Flop  |  958 |       |           |       |
| Unique Control Sets                       |  159 |       |           |       |
+-------------------------------------------+------+-------+-----------+-------+
* Note: Review the Control Sets Report for more information regarding control sets.

3. Memory
---------

+-------------------+------+-------+-----------+-------+
|     Site Type     | Used | Fixed | Available | Util% |
+-------------------+------+-------+-----------+-------+
| Block RAM Tile    | 29.5 |     0 |        50 | 59.00 |
|   RAMB36/FIFO*    |   17 |     0 |        50 | 34.00 |
|     RAMB36E1 only |   17 |       |           |       |
|   RAMB18          |   25 |     0 |       100 | 25.00 |
|     RAMB18E1 only |   25 |       |           |       |
+-------------------+------+-------+-----------+-------+
* Note: Each Block RAM Tile only has one FIFO logic available and therefore can accommodate only one FIFO36E1 or one FIFO18E1. However, if a FIFO18E1 occupies a Block RAM Tile, that tile can still accommodate a RAMB18E1

4. DSP
------

+----------------+------+-------+-----------+-------+
|    Site Type   | Used | Fixed | Available | Util% |
+----------------+------+-------+-----------+-------+
| DSPs           |    4 |     0 |        90 |  4.44 |
|   DSP48E1 only |    4 |       |           |       |
+----------------+------+-------+-----------+-------+
````

enjoy-digital, 2019-10-07:
Arty BaseSoC (DDR3, Vexriscv) with Yosys synthesis / Vivado P&R:
````
1. Slice Logic
--------------

+----------------------------+------+-------+-----------+-------+
|          Site Type         | Used | Fixed | Available | Util% |
+----------------------------+------+-------+-----------+-------+
| Slice LUTs                 | 9582 |     0 |     20800 | 46.07 |
|   LUT as Logic             | 9086 |     0 |     20800 | 43.68 |
|   LUT as Memory            |  496 |     0 |      9600 |  5.17 |
|     LUT as Distributed RAM |  496 |     0 |           |       |
|     LUT as Shift Register  |    0 |     0 |           |       |
| Slice Registers            | 3768 |     0 |     41600 |  9.06 |
|   Register as Flip Flop    | 3768 |     0 |     41600 |  9.06 |
|   Register as Latch        |    0 |     0 |     41600 |  0.00 |
| F7 Muxes                   |  740 |     0 |     16300 |  4.54 |
| F8 Muxes                   |  143 |     0 |      8150 |  1.75 |
+----------------------------+------+-------+-----------+-------+

1.1 Summary of Registers by Type
--------------------------------

+-------+--------------+-------------+--------------+
| Total | Clock Enable | Synchronous | Asynchronous |
+-------+--------------+-------------+--------------+
| 0     |            _ |           - |            - |
| 0     |            _ |           - |          Set |
| 0     |            _ |           - |        Reset |
| 0     |            _ |         Set |            - |
| 0     |            _ |       Reset |            - |
| 0     |          Yes |           - |            - |
| 4     |          Yes |           - |          Set |
| 0     |          Yes |           - |        Reset |
| 0     |          Yes |         Set |            - |
| 3764  |          Yes |       Reset |            - |
+-------+--------------+-------------+--------------+

2. Slice Logic Distribution
---------------------------

+-------------------------------------------+------+-------+-----------+-------+
|                 Site Type                 | Used | Fixed | Available | Util% |
+-------------------------------------------+------+-------+-----------+-------+
| Slice                                     | 2683 |     0 |      8150 | 32.92 |
|   SLICEL                                  | 1883 |     0 |           |       |
|   SLICEM                                  |  800 |     0 |           |       |
| LUT as Logic                              | 9086 |     0 |     20800 | 43.68 |
|   using O5 output only                    |    4 |       |           |       |
|   using O6 output only                    | 8413 |       |           |       |
|   using O5 and O6                         |  669 |       |           |       |
| LUT as Memory                             |  496 |     0 |      9600 |  5.17 |
|   LUT as Distributed RAM                  |  496 |     0 |           |       |
|     using O5 output only                  |    0 |       |           |       |
|     using O6 output only                  |  304 |       |           |       |
|     using O5 and O6                       |  192 |       |           |       |
|   LUT as Shift Register                   |    0 |     0 |           |       |
| LUT Flip Flop Pairs                       | 2547 |     0 |     20800 | 12.25 |
|   fully used LUT-FF pairs                 |    9 |       |           |       |
|   LUT-FF pairs with one unused LUT output | 2347 |       |           |       |
|   LUT-FF pairs with one unused Flip Flop  | 2470 |       |           |       |
| Unique Control Sets                       |  203 |       |           |       |
+-------------------------------------------+------+-------+-----------+-------+
* Note: Review the Control Sets Report for more information regarding control sets.

3. Memory
---------

+-------------------+------+-------+-----------+-------+
|     Site Type     | Used | Fixed | Available | Util% |
+-------------------+------+-------+-----------+-------+
| Block RAM Tile    | 27.5 |     0 |        50 | 55.00 |
|   RAMB36/FIFO*    |   17 |     0 |        50 | 34.00 |
|     RAMB36E1 only |   17 |       |           |       |
|   RAMB18          |   21 |     0 |       100 | 21.00 |
|     RAMB18E1 only |   21 |       |           |       |
+-------------------+------+-------+-----------+-------+
* Note: Each Block RAM Tile only has one FIFO logic available and therefore can accommodate only one FIFO36E1 or one FIFO18E1. However, if a FIFO18E1 occupies a Block RAM Tile, that tile can still accommodate a RAMB18E1

4. DSP
------

+-----------+------+-------+-----------+-------+
| Site Type | Used | Fixed | Available | Util% |
+-----------+------+-------+-----------+-------+
| DSPs      |    0 |     0 |        90 |  0.00 |
+-----------+------+-------+-----------+-------+
````
The DSPs are not infered and it probably explains the large difference for the Slice LUTs.

daveshah1, 2019-10-07:
DSP inference was added to synth_xilinx last week. If Yosys is up to date and DSPs are still not being inferred, please file an issue.

enjoy-digital, 2019-10-07:
@daveshah1: thanks, i'm not using the last version of Yosys. I'll update.

enjoy-digital, 2019-10-07:
I think we can now close this issue.

closed 2019-10-07 by enjoy-digital