enjoy-digital/litex #133: Land minimal / ice40 optimized version of vexriscv

mithro, 2018-12-16:
Currently VexRISCV only has a `full` and `debug` variants set up. It would be good to have `lite` and `minimal` versions added too.

I believe @cr1901 was looking into this.

enjoy-digital, 2018-12-16:
I already added it with https://github.com/enjoy-digital/litex/commit/35155e517284895a34dfa77cf8ad4c1f8c71938a. And then did a bit of refactoring with https://github.com/enjoy-digital/litex/commit/e9f10492007fab2a9b26cbb3539fc09f4e587f3d. Can you look at tell me if that's what you want?

mithro, 2018-12-16:
Looks like I haven't updated litex in a while. Doing a version roll and seeing if this works for us.

xobs, 2018-12-19:
FWIW, I managed to synthesize `minimal` on a ICE40UP5K.  It uses 400 more gates than an lm32.

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
 SoC BIOS / CPU: VexRiscv / 16MHz

(c) Copyright 2012-2018 Enjoy-Digital
(c) Copyright 2007-2018 M-Labs Limited
Built Dec 19 2018 14:37:07

Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
?!?!>
```

enjoy-digital, 2019-01-27:
I think we can close this now, this works and can be tested easily with:
- litex_sim --cpu-type=vexriscv --cpu-variant=min
- litex_sim --cpu-type=vexriscv --cpu-variant=min_debug
- litex_sim --cpu-type=vexriscv --cpu-variant=lite
- litex_sim --cpu-type=vexriscv --cpu-variant=lite_debug
- litex_sim --cpu-type=vexriscv --cpu-variant=std
- litex_sim --cpu-type=vexriscv --cpu-variant=std_debug

closed 2019-01-27 by enjoy-digital