enjoy-digital/litex #134: Land linux optimized version of vexriscv

mithro, 2018-12-16:
@antmicro is working on 32bit Linux on RISCV. It would be good to have the vexriscv variant which supports linux in the same way there is a or1k variant which supports linux.

enjoy-digital, 2019-01-27:
Just listing repositories used by @antmicro for the Polarfire Avalanche Vexriscv linux demo:
- https://github.com/antmicro/litex-linux-readme
- https://github.com/antmicro/litex-rv32-linux-system
- https://github.com/antmicro/avalanche
- https://github.com/antmicro/litex/commits/linux
- https://github.com/antmicro/VexRiscv/tree/Supervisor

It will be interesting to understand the change, merge that back.

enjoy-digital, 2019-04-28:
VexRiscv now have have a real MMU: https://github.com/SpinalHDL/VexRiscv/issues/60
I started testing and integrating it here: https://github.com/enjoy-digital/linux-on-litex-vexriscv

enjoy-digital, 2019-05-13:
Linux variant of VexRiscv is now integrated in LiteX and used to build https://github.com/enjoy-digital/linux-on-litex-vexriscv.

closed 2019-05-13 by enjoy-digital