enjoy-digital/litex #139: Enable link time optimisation in builds

mithro, 2018-12-17:
Link time optimisation seems to be resulting in much smaller binaries as any unused elements can be totally removed.

There are however some issues to make this work;
 - [ ] Have to use a modern gcc
 - [ ] Use GCC for linking
 - [ ] Have to mark a couple of functions like the ISR as actually used.
 - [ ] Have to link against libgcc

mithro, 2018-12-17:
I have a WIP branch at https://github.com/mithro/litex/tree/enable-lto?files=1

mithro, 2018-12-17:
It is unclear why we have to link against libgcc when the symbols should be provided by `libcompiler_rt`. It seems to be because GCC is generating function calls itself (rather than being in the code) which cause lto to not understand that the symbols are used.

enjoy-digital, 2020-02-25:
This has been integrated with https://github.com/enjoy-digital/litex/pull/401. For now it's enabled with for all processors except LM32. If this is causing issue with others configurations, we'll add a parameter be able to disable LTO when needed.

enjoy-digital, 2020-03-11:
LTO has been reverted in https://github.com/enjoy-digital/litex/commit/979f98ea3120dbba208c2b0b044ce37294653814 since seems to cause subtle issues on some configurations. We'll have to test it more before enabling it.

enjoy-digital, 2020-03-11:
cc @mithro @mateusz-holenko.

ewenmcneill, 2020-03-12:
In case it helps, the link issue I had (in https://github.com/enjoy-digital/litex/issues/417) with `litex-buildenv` was with the "stub firmware" in `litex-buildenv`, which uses the auto-generated `litex` compiler/linker/flags/etc, but has its own `Makefile`.  So (copying from my comment on https://github.com/enjoy-digital/litex/issues/417#issuecomment-597940082), " updating the definition of $(LD) in `common.mak` instead of changing it locally in just the BIOS `Makefile`" might make these changes more robust (/consistent) for anything else that uses the `litex` generated compiler/linker/flags, etc.  (It looks like that `Makefile` suffered from trying to use `ld` with flags intended for `gcc`, via `$(CPUFLAGS)`, on `lm32`....)
  
Also (as I said in the same comment) maybe having a build time option to "force LTO off" would help anyone thinking their target is suffering from bad LTO handling.  

In general I think it'd be a great idea to have LTO available as a build option, and probably even on by default. 

Ewen

/cc @mithro @mateusz-holenko

mithro, 2020-03-12:
Having LTO on by default with a way to easily turn it off seems like a good plan. If LTO is off by default, it will just end up bit rotting until it no longer works.

enjoy-digital, 2020-03-12:
I'm also ok with that, we just need to work a bit more on it before integrating it.

DurandA, 2020-11-03:
I just found this issue, should I close https://github.com/enjoy-digital/litex/issues/682 which is a duplicate of this?

>     * [ ]  Have to link against libgcc

This is not necessary. It was required because libcompiler_rt was optimized out.

ewenmcneill, 2020-11-05:
> should I close #682 which is a duplicate of this?

@DurandA It looks to me like #682 is much more active now.  I'd probably lean towards closing *this* issue (which was already closed once, then reopened when the patch was reverted), with a comment like "LTO (re-)implementation work moved to #682"...

(Looks like @mithro or @enjoy-digital digital would have to be the ones to close this specific older issue.)

Ewen

mithro, 2020-11-05:
Let's keep #682 and close this one.

closed 2020-11-05 by mithro