enjoy-digital/litex #145: crt0 doesn't copy data section into RAM?

mithro, 2019-01-23:
There seems to be some confusion around if the data section should be copied to RAM.

If the program is executing from a read only location (ROM, spiflash, etc) - then the crt0 needs to copy the data section into RAM.
If the program is executing from a read/write location (SRAM, DDR, etc) - then the crt0 doesn't need to do anything (as the data section can be modified in place).

Both [`picorv32`](https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/libbase/crt0-picorv32.S#L197-L209) and [`lm32`](https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/libbase/crt0-lm32.S#L115-L131) have a `EXECUTE_IN_PLACE` define which does the data section copy.

However, [`vexriscv`](https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/libbase/crt0-vexriscv.S) and [`or1k`](https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/libbase/crt0-or1k.S) and [`minerva`](https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/libbase/crt0-minerva.S) are missing this.

Naming this `EXECUTE_IN_PLACE` is kind of misleading, as the reason the copy needs to occur is because the place it is executing from is read only.

mithro, 2019-01-23:
@xobs @cr1901 - FYI.

cr1901, 2019-01-23:
It's missing for @vexriscv, @or1k and @minerva because I haven't implemented it yet :).

>the reason the copy needs to occur is because the place it is executing from is read only.
I think it's more subtle than that. You can in principle have a read only code section in SPI flash, but e.g. SRAM is initialized during bitstream loading. No copy is needed in that case either.

I mainly used `EXECUTE_IN_PLACE` as a shorthand for "code and data do not live in the same section of address space from the CPU's point-of-view." Open to suggestions to change it (as well as `crt0-xip.S`/`crt0-ctr.S`).

mithro, 2019-01-23:
I would suggest that we need something more like `EXECUTING_FROM_RO_REGION` and `EXECUTING_FROM_RW_REGION`?

ewenmcneill, 2019-03-09:
I randomly found this issue down a chain of links; some thoughts off the top of my head (as I remember discussing this when the first TinyFPGA BX MicroPython bringup was happening, which I think led to the existing macro name...):

*   The two cases are basically "everything copied into read/write RAM" (non-XIP) and "only `.data` is copied into read/write RAM" (XIP)

*   That "everything" copy is basically implicit in copying the bitstream into the FPGA, so it's the "only `.data` is copied" case which is special, because `.data` is not in the bitstream (to go into RAM) and thus needs handling later.

*   Perhaps just calling the compile flag something like `COPY_DATA_FROM_RO` would be accurate, still relatively short, and provide a hint to what was happening.  Or `DATA_IN_BITSTREAM` `DATA_NOT_IN_BITSTREAM`?

*   The logic for setting `COPY_DATA_FROM_RO` (or `DATA_NOT_IN_BITSTREAM`)  is currently the same as the existing"`EXECUTE_IN_PLACE`" (which IIRC sets some other things in the linker, and might need to be retained anyway).

Ewen

mithro, 2019-03-09:
The split between the source being read/write or read only does sound like an important point.

FYI - The firmware in bitstream could either be read/write (BRAM emulating RAM with preloaded content) or read only (BRAM emulating ROM). We mostly use the second mode but on the smaller devices the first mode probably makes a lot more sense.

enjoy-digital, 2020-06-23:
This has been fixed with https://github.com/enjoy-digital/litex/issues/566 and we no longer distinguish the case in crt0. Even if the firmware is executed from a read/write location, it's better to also copy the data section to RAM to allow correct reset of the system.

closed 2020-06-23 by enjoy-digital