enjoy-digital/litex #146: Problem with fpga 101 lesson/lab003

Lexuil, 2019-01-28:
A months ago, I didn't have this problem. But i update litex and when i try to execute `python3 base.py` I get this:

```
Traceback (most recent call last):
  File "base.py", line 6, in <module>
    from litex.soc.integration.soc_core import *
  File "/usr/local/lib/python3.5/dist-packages/litex-0.2.dev0-py3.5.egg/litex/soc/integration/__init__.py", line 2, in <module>
    from litex.soc.integration.soc_sdram import SoCSDRAM
  File "/usr/local/lib/python3.5/dist-packages/litex-0.2.dev0-py3.5.egg/litex/soc/integration/soc_sdram.py", line 8, in <module>
    from litedram.frontend import crossbar
ImportError: cannot import name 'crossbar'
```

https://github.com/enjoy-digital/fpga_101/tree/master/lab003

Lexuil, 2019-01-30:
I solve it.

I execute `sudo python3 setup.py develop` in litex root file.

closed 2019-01-30 by Lexuil

enjoy-digital, 2019-01-30:
Thanks, yes, it's probably that LiteX was not up to date.