enjoy-digital/litex #160: Allow litex_server to bind to other interfaces

xobs, 2019-04-13:
The `litex_server` binary is incredibly useful for bridging Wishbone to Ethernet.  It is some of the magic that makes Litex work.

I'm running `litex_server` on a Raspberry Pi, and I would like to run `openocd` and/or `devmem2_litex` on my development system.  This requires that I bind `litex_server` to "0.0.0.0" instead of to "localhost".

Right now I manually modify `litex_server`, however this change is unsuitable for upstreaming because it is insecure.

It would be nice if there was a way to add an option (command-line switch, environment variable, or something else) to `litex_server` to enable this functionality.

enjoy-digital, 2019-04-14:
Hi @xobs,

thanks, yes this is a good idea. Can you check if https://github.com/enjoy-digital/litex/pull/161 is what you want? I also changed the way arguments are passed to allow passing the bind address and ease adding future arguments.

xobs, 2019-04-14:
That's almost exactly what I was looking for!

The only other issue I've run into is `TCP_WAIT`, where an unclean termination means that port 1234 is unavailable for 60 seconds.  This causes `litex_server` to bind to port 1235, which is not very useful because my openocd connection is explicitly trying to connect to port 1234.

Is it possible to add a `--port` option that would remove the `for i in range(32):` loop?

enjoy-digital, 2019-04-14:
I just updated to implement that, is it fine?

xobs, 2019-04-14:
It does, thanks!

Do you need the port-walking behavior at all, or is that a holdover from something else?

enjoy-digital, 2019-04-14:
Sorry, i'm not sure to understand for the port-walking, i already removed the `for i in range(32)`, is it what you are talking about?

xobs, 2019-04-14:
Yes, I'm sorry.  My question was: Why was `for i in range(32)` there to begin with?  Is it because `litex_server` needed to pick a port, and sometimes that port was in use?

enjoy-digital, 2019-04-14:
Yes that was to run run multiple instances of `litex_server` on the same machine. That's fine and cleaner to use the new `--bind-port` argument. Do you know if we can do something to get the the port back before waiting 60s in case of unclean termination? (or to reduce this timeout?)

xobs, 2019-04-14:
It may be possible to set `SO_REUSEADDR` (by setting `self.socket.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)`) to reuse that port.

enjoy-digital, 2019-04-14:
Thanks i'll try it tomorrow and will also probably merge https://github.com/enjoy-digital/litex/pull/161.

enjoy-digital, 2019-04-15:
https://github.com/enjoy-digital/litex/pull/161 is merged, thanks for the ideas/suggestions.

closed 2019-04-15 by enjoy-digital