enjoy-digital/litex #177: Import the Taiga high performance RISC-V core

mithro, 2019-05-04:
[Taiga](https://gitlab.com/sfu-rcl/Taiga) is a 32-bit RISC-V processor designed for running Linux and SMP type things on FPGAs. Taiga is licensed under the [Apache License, Version 2.0](https://www.apache.org/licenses/LICENSE-2.0)

Currently the only Linux supporting RISC-V processor we have is VexRISCV. It would be good to have a second RISC-V processor which targets high performance and Linux.

See the following description;
> Taiga is a 32-bit RISC-V processor designed for FPGAs supporting the Multiply/Divide and Atomic extensions (RV32IMA).  The processor is written in SystemVerilog and has been designed to be both highly extensible and highly configurable.
> The pipeline has been designed to support parallel, variable-latency execution units and to readily support the inclusion of new execution units.
> [image]

mithro, 2019-05-04:
FYI Here are a collection of links related to Linux on RISC-V;
 * https://github.com/enjoy-digital/linux-on-litex-vexriscv
 * https://github.com/antmicro/litex-linux-readme
 * https://github.com/SpinalHDL/VexRiscv/issues/60 - Linux on VexRiscv
 * https://github.com/antmicro/litex-linux-riscv
 * https://github.com/antmicro/litex-rv32-linux-system
 * https://github.com/SymbiFlow/ideas/issues/35
 * https://github.com/SymbiFlow/ideas/issues/34

kamejoko80, 2020-01-09:
I’m interested in LiteX with multi core support. Do you have a plan or progress of it?

mithro, 2020-01-09:
@kamejoko80 - This issue is about adding the Taiga core to LiteX rather than multicore support in general (which we plan to do using LiteX with VexRISCV multicore in 2020 -- the first step is to add multicore support to VexRISCV see https://github.com/SpinalHDL/VexRiscv/issues/85 ).

kamejoko80, 2020-01-10:
@mithro 
That's great! I'm looking forward to hearing from your announcement.
Thank you.

enjoy-digital, 2020-05-19:
There are no ongoing efforts to import/support this core and now have others RISC-V linux capable alternatives (Rocket, BlackParrot), so i'm closing but please re-open if someone is willing to put efforts into this.

closed 2020-05-19 by enjoy-digital