enjoy-digital/litex #180: Duplicate constraint raises wrong exception

TomKeddie, 2019-05-10:
At https://github.com/enjoy-digital/litex/blob/master/litex/build/xilinx/vivado.py#L254 it raises ValueError if a constraint exists.  I think it should raise ConstraintError so I can be caught at https://github.com/enjoy-digital/litex/blob/master/litex/build/generic_platform.py#L314

I'm still feeling my way around litex so perhaps I'm missing something here.

  File "/home/tom/git/im-tomu/foboot/hw/deps/litex/litex/build/xilinx/platform.py", line 48, in add_period_constraint
    self.toolchain.add_period_constraint(self, clk, period)
  File "/home/tom/git/im-tomu/foboot/hw/deps/litex/litex/build/xilinx/vivado.py", line 254, in add_period_constraint
    raise ValueError("A period constraint already exists")
ValueError: A period constraint already exists

enjoy-digital, 2019-05-10:
Hi @TomKeddie,

can you share the code you are executing to have this behaviour? Things can maybe be improved but i'd first like to see in which situation you are seeing this.

Thanks,

Florent

TomKeddie, 2019-05-11:
Sorry to bother you, there were multiple conflicting constraints on the clock.

closed 2019-05-11 by TomKeddie