enjoy-digital/litex #181: Minerva CPU support

mateusz-holenko, 2019-05-15:
I was trying to build LiteX BIOS for Minerva CPU and noticed a strange thing:

There is a minerva-specific code in https://github.com/enjoy-digital/litex/blob/8f377307d8b257dbd1e74805c8519abdc04e884d/litex/soc/software/include/base/irq.h#L111, but it looks like a copy-paste from VexRiscv.

This file: https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/include/base/csr-defs.h defines CSRs, but the values are VexRiscv-specific.

Here: https://github.com/lambdaconcept/minerva/blob/master/minerva/isa.py#L65 I can see that minerva supports both standard RISC-V CSRs (mie, mip, mstatus) and some "µarch specific": IRQ_MASK and IRQ_PENDING, but their values are different than those for VexRiscv.

What is the status of Minerva support? Is this supposed to work?

mithro, 2019-05-15:
FYI The minvera repository can be found at https://github.com/lambdaconcept/minerva

enjoy-digital, 2019-05-15:
Hi @mateusz-holenko,

Minerva registers were based on VexRiscv's ones, and are still using the old registers VexRiscv's was using before https://github.com/enjoy-digital/litex/issues/87. 

I just fixed that with https://github.com/enjoy-digital/litex/commit/1ea22d49b7d0b0b200c9599836458070c31a5d4b and tried with:
````
lxsim --cpu-type=minerva
````

LiteX is still not using the last version of Minerva since there was a issue to fix with nMigen, but we should switch to it soon.

enjoy-digital, 2019-05-20:
With https://github.com/enjoy-digital/litex/commit/10670e22acf82a3ea9259bf54e74ea095c6c8851 we are now supporting the latest version of Minerva.

closed 2019-05-20 by enjoy-digital