enjoy-digital/litex #205: TFTP transfer bug

tweakoz, 2019-06-27:
I was experiencing strange TFTP transfer issues. I diagnosed it in my chainloader (where I made a copy of the tftp/udp code). The culprit was documented here:
https://github.com/tweakoz/litex-chainloader/blob/master/tftp.cpp#L136
I would submit a fix and pull request, but I would not have time to test all scenarios, atm...
The fix seemed to work on my chainloader - so the same approach might work here...

enjoy-digital, 2019-06-27:
Thanks for the bug identification and litex-chainloader. I'll look at your code, if you also idenfified a fix, you can create a pull request, i'll review it and will do more tests.

mithro, 2019-06-27:
Should litex-chainloader become part of LiteX-Hub?

tweakoz, 2019-06-28:
As is, making it part of Litex proper would pollute the bios with c++. I ported all Litex code to embedded style c++17 in the process of getting it working - it helped me understand it better. Not that I personally would mind c++ in the bios, but the c-purists out there might... It's a subjective thing... could always port it back to c if desired.

enjoy-digital, 2019-06-28:
We should probably keep the bios in C, but the ideas with the chainloader are really interesting and it would be useful to have a  these features. i'll first look at the bug, then see about the chainloader. (I think @mithro was suggesting to integrate your repo to https://github.com/litex-hub, which i would also agree if we don't end up with a similar solution directly integrated in LiteX)

enjoy-digital, 2019-07-27:
@tweakoz, sorry for the delay, looking at: http://www.faqs.org/rfcs/rfc1350.html and https://en.wikipedia.org/wiki/Trivial_File_Transfer_Protocol it seems we are implementing the protocol correctly:

> The final DATA packet must contain less than a full-sized block of data to signal that it is the last. If the size of the transferred file is an exact multiple of the block-size, the source sends a final DATA packet containing 0 bytes of data.

So the case where filesize is an integer multiple of BLOCK_SIZE is already covered by the actual code: the server should send a final packet with 0 bytes of data, was your TFTP server doing this?

enjoy-digital, 2019-08-02:
@tweakoz: i'm closing regarding my previous analysis and since i think we are doing this correctly. Please re-open if you still think there is an issue.

closed 2019-08-02 by enjoy-digital