enjoy-digital/litex #211: Accessing Memory special from firmware

DurandA, 2019-07-05:
I am trying to access the content of a `Memory` from the firmware using the `Identifier` core:
https://github.com/enjoy-digital/litex/blob/6b82f23ce191af914b5a0a57feebe6873915aa2a/litex/soc/cores/identifier.py#L7-L14

I added an identifier to my SoC definition using`self.submodules.identifier = Identifier("HelloWorld")` and `CSR_IDENTIFIER_MEM_BASE` was generated in *csr.h*.

Since the identifier's Memory is null-terminated, I expected to be able to read its content like a string literal in C:
```
puts((char *)CSR_IDENTIFIER_MEM_BASE);
```

However, only the first character "H" is outputted. I think this is due to the memory layout and that the next memory address contains 0 (`\0`).

What is the correct way to access contiguous memory data from the firmware?

enjoy-digital, 2019-07-05:
It would maybe work with a `csr_data_width` of 32. Here is how you can do it otherwise: https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/libbase/id.c

DurandA, 2019-07-06:
Using a csr_data_width of 32 does not change the behavior with my SoC definition. Thank you for sharing *libbase/id.c*.

Would you mind explaining the following line?
https://github.com/enjoy-digital/litex/blob/282ae9635465fca96f0c85da3eaa9a6f29197350/litex/soc/software/libbase/id.c#L13

Many thanks!

enjoy-digital, 2019-07-06:
Sorry, the software was not up to date with the gateware. Previously, the first byte was the length, we now just use a fixed size of 256, i updated the code with https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/libbase/id.c. For this memory located on the CSR bus, i would recommend doing the read as done in `get_ident`, for the others memory located on the Wishbone bus, you should be able to do what you were suggesting.

DurandA, 2019-07-07:
Thank you for your time!

DurandA, 2019-07-08:
https://github.com/enjoy-digital/litex/blob/f4770219fa87640a61328ec92a9b2ceef40efe03/litex/soc/software/libbase/id.c#L12-L13

While it seems to work with my configuration, can't it trigger illegal exceptions when Memory width is < 256? Are CSR addresses always valid no matter what?

enjoy-digital, 2019-07-08:
The CSR addresses are indeed always valid, this will not cause illegal exceptions.

closed 2019-07-08 by DurandA