enjoy-digital/litex #212: Overlapped memory regions fail silently after bitstream generation

xobs, 2019-07-08:
The `sram` region has moved, which can cause some instability in designs that make assumptions about memory regions.

In particular, memory used to be located at offset `0x10000000`.  Therefore, it is reasonable to have something like this in a design:

```python
        from litex.soc.cores import up5kspram
        spram_size = 128*1024
        self.submodules.spram = up5kspram.Up5kSPRAM(size=spram_size)
        self.register_mem("sram", 0x10000000, self.spram.bus, spram_size)
```

However, this region overlaps with the new offset of the `csr` region, which is at 0x02000000-0x12000000.

This design will successfully generate a bitstream, however any accesses to the `sram` region will be silently ignored.

mithro, 2019-07-08:
Agreed that overlapping memory regions should probably produce an error?

FYI - I think you want the following however?
```patch
diff --git a/targets/icebreaker/base.py b/targets/icebreaker/base.py
index 580130d..6f31db0 100644
--- a/targets/icebreaker/base.py
+++ b/targets/icebreaker/base.py
@@ -118,7 +118,7 @@ class BaseSoC(SoCCore):
         # SPRAM- UP5K has single port RAM, might as well use it as SRAM to
         # free up scarce block RAM.
         self.submodules.spram = ice40.SPRAM(size=128*1024)
-        self.register_mem("sram", 0x10000000, self.spram.bus, 0x20000)
+        self.register_mem("sram", self.mem_map["sram"], self.spram.bus, 0x20000)
 
         # We don't have a DRAM, so use the remaining SPI flash for user
         # program.
diff --git a/targets/upduino_v1/base.py b/targets/upduino_v1/base.py
```

enjoy-digital, 2019-07-08:
Thanks for reporting, this is fixed with https://github.com/enjoy-digital/litex/commit/21a5aaa4a62d31568de992e650fd9daa2899b8db.

commit 21a5aaa4 (2019-07-08): soc_core: declare csr address size when registering csr, fixes #212
closed 2019-07-08 by enjoy-digital