enjoy-digital/litex #213: Use of GPIO:N for Pin declaration is confusing

DurandA, 2019-07-08:
The use of `GPIO:0` and `GPIO:1` for the _tinyfpga_bx_ board is quite confusing:
https://github.com/enjoy-digital/litex/blob/f4770219fa87640a61328ec92a9b2ceef40efe03/litex/boards/platforms/tinyfpga_bx.py#L50-L56

It is exclusive to this board and doesn't seem to match the GPIO order (I would expect `GPIO:0` to be _A2_):
https://github.com/enjoy-digital/litex/blob/f4770219fa87640a61328ec92a9b2ceef40efe03/litex/boards/platforms/tinyfpga_bx.py#L45

enjoy-digital, 2019-07-08:
For this board, since there is no serial by default, we define the pins of the connectors (GPIO and EXTRA), and then use these connectors to add a default serial. `GPIO:0` will be converted to A2, 'GPIO:1' will be converted to A1. The aim is to simplify things by defining the serial against a connector number instead of the FPGA pins.

DurandA, 2019-07-08:
For some reason, the _tx_ pin (`GPIO:0`) maps to _C2_ instead of _A2_. I am unsure about _rx_.

I simply used [`platform.add_extension(tinyfpga_bx.serial)`](https://github.com/DurandA/tinyfpga-litex/blob/338a76d71a8d4e6b57054793d2899fb7b2cc6bce/tinyfpga_litex.py#L42) to add the serial peripheral.

DurandA, 2019-07-08:
Here is the generated top.pcf:
```
set_io serial_tx C2
set_io serial_rx B1
set_io clk16 B2
set_io spiflash_cs_n F7
set_io spiflash_clk G7
set_io spiflash_mosi G6
set_io spiflash_miso H7
set_io spiflash_wp H4
set_io spiflash_hold J8
set_io user_led0 B3
```

I rebuilt the project to make sure that the file is up to date.

enjoy-digital, 2019-07-08:
OK thakns, then i'll look at that.

DurandA, 2019-07-08:
Not directly related, but the `litex_boards.partner` package [cannot be imported](https://github.com/litex-hub/litex-boards/issues/6`) (including the `tinyfpga_bx` target) until [this pull request](https://github.com/litex-hub/litex-boards/pull/5) is merged.

DurandA, 2019-07-08:
Complete facepalm from me as I edited my _litex_boards_ package. :no_mouth:
`GPIO:0` and `GPIO:1` are indeed behaving correctly.

Please accept my apologies.

closed 2019-07-08 by DurandA

enjoy-digital, 2019-07-08:
No problem, don't worry :) (you raised others valid issues while trying to understand the code)