enjoy-digital/litex #215: Improve SpiFlash

enjoy-digital, 2019-07-10:
- [ ] make dummy parameter dynamically configurable. (for some boards, SPI Flash chip can vary between revisions (Arty for example), adding dynamic dummy support would allow using the same bistream for the different revisions. This would also speedup finding dummy parameter for new boards: mw/mr commands frome the bios could be used instead of bistream iterations.
- [ ] add Dual/Quad support to bitbang mode.
- [ ] remove duplication between SpiFlashDualQuad/SpiFlashSingle.
- [ ] add test.

mithro, 2019-07-10:
Maybe we should create a "LiteSPI module"?

mithro, 2019-07-10:
There is some WIP stuff related to SPI in the pull request at https://github.com/enjoy-digital/litex/pull/203

enjoy-digital, 2019-07-10:
I think the complexity in SPI Flash is mostly handling the number of manufacturers/configurations, not really the SPI Flash core in itself. Creating a an external module could be useful if others projects want to reuse these definitions, but maybe it should be named LiteSPIFlash then.

mithro, 2019-07-11:
One thing to check is if we can just use the BIOS to set the config register which controls the dummy bits in use to match what the gateware is expecting....

xobs, 2019-07-17:
How difficult is it to mux the SPI flash module with the `SB_SPI` hard macro?  Is there an easy way to insert a mux to allow me to disconnect the `Pin`s from this and re-use them in order to get a higher-performance write using hard IP?

enjoy-digital, 2020-04-10:
SPI Flash support is currently being re-written in LiteSPI (https://github.com/litex-hub/litespi). The new core will cover all features that were planned.

closed 2020-04-10 by enjoy-digital

mithro, 2020-04-10:
@xobs - Can you log an issue about the SB_SPI hard macro on the litespi repository?