enjoy-digital/litex #220: Memory change adds 300 LCs

xobs, 2019-07-22:
The changes in 740629ba53944fc04fca79e433508a67cfe0f02f have caused an additional 300 LCs to be allocated, exceeding the budget on an ICE40.

Before 740629ba53944fc04fca79e433508a67cfe0f02f (i.e. at f49d0fe6c9281c36307aa0b9f51050aa9d6ef152):

```
Info: Device utilisation:
Info:            ICESTORM_LC:  5120/ 5280    96%
Info:           ICESTORM_RAM:    29/   30    96%
Info:                  SB_IO:    14/   96    14%
Info:                  SB_GB:     8/    8   100%
Info:           ICESTORM_PLL:     1/    1   100%
Info:            SB_WARMBOOT:     1/    1   100%
Info:           ICESTORM_DSP:     0/    8     0%
Info:         ICESTORM_HFOSC:     0/    1     0%
Info:         ICESTORM_LFOSC:     0/    1     0%
Info:                 SB_I2C:     0/    2     0%
Info:                 SB_SPI:     0/    2     0%
Info:                 IO_I3C:     0/    2     0%
Info:            SB_LEDDA_IP:     1/    1   100%
Info:            SB_RGBA_DRV:     1/    1   100%
Info:         ICESTORM_SPRAM:     4/    4   100%
```

With no other changes, commit 740629ba53944fc04fca79e433508a67cfe0f02f is now over-budget:

```
Info: Device utilisation:
Info:            ICESTORM_LC:  5439/ 5280   103%
Info:           ICESTORM_RAM:    29/   30    96%
Info:                  SB_IO:    14/   96    14%
Info:                  SB_GB:     8/    8   100%
Info:           ICESTORM_PLL:     1/    1   100%
Info:            SB_WARMBOOT:     1/    1   100%
Info:           ICESTORM_DSP:     0/    8     0%
Info:         ICESTORM_HFOSC:     0/    1     0%
Info:         ICESTORM_LFOSC:     0/    1     0%
Info:                 SB_I2C:     0/    2     0%
Info:                 SB_SPI:     0/    2     0%
Info:                 IO_I3C:     0/    2     0%
Info:            SB_LEDDA_IP:     1/    1   100%
Info:            SB_RGBA_DRV:     1/    1   100%
Info:         ICESTORM_SPRAM:     4/    4   100%
```

enjoy-digital, 2019-07-22:
Thanks for the feedback, https://github.com/enjoy-digital/litex/commit/41eb21b343be01c7fcf8e37345a27df82473fc2c should make the new decoder alsmost equivalent to the old one in your case (256MB aligned addresses). I did a synthesis test with Foboot to verify resource usage, can you test it?

xobs, 2019-07-22:
That change does improve things by quite a bit:

```
Info: Device utilisation:
Info:            ICESTORM_LC:  5143/ 5280    97%
Info:           ICESTORM_RAM:    29/   30    96%
Info:                  SB_IO:    14/   96    14%
Info:                  SB_GB:     8/    8   100%
Info:           ICESTORM_PLL:     1/    1   100%
Info:            SB_WARMBOOT:     1/    1   100%
Info:           ICESTORM_DSP:     0/    8     0%
Info:         ICESTORM_HFOSC:     0/    1     0%
Info:         ICESTORM_LFOSC:     0/    1     0%
Info:                 SB_I2C:     0/    2     0%
Info:                 SB_SPI:     0/    2     0%
Info:                 IO_I3C:     0/    2     0%
Info:            SB_LEDDA_IP:     1/    1   100%
Info:            SB_RGBA_DRV:     1/    1   100%
Info:         ICESTORM_SPRAM:     4/    4   100%
```

The resulting design seems stable, and still seems capable of operating.  Looks good to me!

closed 2019-07-22 by xobs

enjoy-digital, 2019-07-22:
Great, thanks for the quick feedback :)