enjoy-digital/litex #225: how to define constraints?

AnttiLukats, 2019-08-01:
for BML s7-mini it is needed to add 

        self.add_platform_command("set_property CLOCK_DEDICATED_ROUTE FALSE [get_nets clk100_IBUF]")

this WORKS but does throw error during compile, so it is possible applied at wrong time, where and how should such contstraints be added?

it is not a showstopper as full compile does succeed, but there is one RED message during the vivado flow

enjoy-digital, 2019-08-01:
The location where you use it should not be important and if it works it means Vivado is able to recognize it (i know it fails without it), can you share the full error message and a minimal design to be able to reproduce?

AnttiLukats, 2019-08-01:
https://github.com/micro-FPGA/litex-boards/blob/master/litex_boards/partner/platforms/s7mini.py

at the END there is "clock dedicated route" line..

it does work as it is, and produce working bit file, but during compilation there is red colored error, would be nice to get rid of the red color..

enjoy-digital, 2019-08-01:
ok, it's probably just the colorama rules that need to be adjusted:
https://github.com/enjoy-digital/litex/blob/master/litex/build/xilinx/common.py#L29-L38

AnttiLukats, 2019-08-01:
I'm not so sure, it looked like there was a warning too much, but the XDC file at the end was correct and everything works, so unless the issue comes back, could be considered not so important.

enjoy-digital, 2019-08-01:
Re-opening, i'll have a look.

enjoy-digital, 2019-08-01:
The RED was due to a critical warning, because clk100_IBUF is not yet existing during synthesis the first time the .xdc is read. I changed the severity of the critical warning to a simple warning with: https://github.com/enjoy-digital/litex/commit/7e0ea0707698b526ee42558fc38b95aa15561bf2
To avoid this warning we would need to use a different .xdc between synthesis and place and route but i'm not sure it's worth doing that vs having these warnings during synthesis.

closed 2019-08-01 by enjoy-digital

AnttiLukats, 2019-08-05:
its a decision to make :) trenz electronic BUILD system has different XDC files they have different naming convention, so some are pulled in earlier than others.. but the system is not used so efficiently, so while the flow allows proper placement of the constraints they still get into wrong file sometimes (manual error then)