enjoy-digital/litex #226: quartus build fails on win32

AnttiLukats, 2019-08-01:
while vivado on windows is supported it seems quartus support does not check for win32 at all?

enjoy-digital, 2019-08-01:
Not sure it has been tested recently on windows machine, have you tested with Quartus define in your PATH?

AnttiLukats, 2019-08-01:
quartus is searched in /opt/

I am sure I can not define anything anywhere if build script has hard coded /opt/

for vivado there it is different, and "win32" is tested before running vivado, also the quartus build does not produce a BAT file as vivado build does, but .sh what would not work on windows..

enjoy-digital, 2019-08-02:
Quartus now uses a .bat file on windows with: https://github.com/enjoy-digital/litex/commit/a496760cb698d3c313853d0f2b27f230a146a20e. It was previously reliying on a linux shell to be installed on windows an this should no longer be the case. Quartus still need to be set in your windows PATH, but i don't think that's a strong requirement.

closed 2019-08-02 by enjoy-digital