enjoy-digital/litex #227: bios build on win32

AnttiLukats, 2019-08-01:
when compiling software there is error:

C:\Python34 -m litex.soc.software.mkmscimg bios.bin --little
process_begin: CreateProcess(NULL, C:\Python34 -m litex.soc.software.mkmscimg bios.bin --little, ...) failed.

where does litex get "C:\Python34 ??

python is in C:\Python37

surprising everything else except the msmscimg thing seem to work..

enjoy-digital, 2019-08-01:
It seems by default `python3` is pointing to  _C:\Python34_ on your machine (LiteX is calling mkmscimg with this https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/common.mak#L4, not _C:\Python34_ directly), can you make sure that in a terminal, python3 is pointing to a valid python3 interpreter?

AnttiLukats, 2019-08-01:
I got it working with extra env setting, what made me wonder was: fusesoc worked with no issues, litex MOSTLY worked as well.. so I guess litex could also work out of box without extra env settings..

anyway, I got litex to build full vexriscv on s7mini board and it worked immediatly, and that compiled on true win32 w7 machine, no linux layer anywhere..

pretty impressed, litex was a bit work on w32 but is now fully working, fusesoc did look like EASY but fails with some weird jinja template not found issue, so litex is leading 1:0 right now

closed 2019-08-01 by AnttiLukats