enjoy-digital/litex #228: too many LE used on Intel FPGA's?

AnttiLukats, 2019-08-02:
LiteX soc compiled for Xilinx has 2100 LUT, while for MAX10 it consumes always over 11,000 LE ?

vexriscv itself takes only 1300 LE, so the LiteX soc TOP consumes 9700 LE ? trying to optimize it to be smaller but it seems to be way off scale on the utilization?

pretty large NIOS SoC designs with SDRAM easily fit MAX10 10M08 with plenty of room left over, simplest smallest LiteX without SDRAM does not fit, seems like not right

daveshah1, 2019-08-02:
I've seen in the past that certain BRAM patterns are accepted by most tools but don't infer BRAM in Quartus. Maybe it is related to that.

AnttiLukats, 2019-08-02:
after a walk did find them, I was just not believing that 32KByte of RAM could be fit into <9000 LE, but actually it is, the ROM is generated from LE's so thats why the LE count explodes, but this HOW come this is not fixed yet? All Altera designs should suffer the same issue I assume?

daveshah1, 2019-08-02:
Looks like LiteX uses this kind of pattern:
```verilog
reg [31:0] mem[0:8191];
reg [12:0] memadr;
always @(posedge sys_clk) begin
	memadr <= basesoc_rom_adr;
end

assign basesoc_rom_dat_r = mem[memadr];

initial begin
	$readmemh("mem.init", mem);
end
```

I suspect Quartus doesn't like the register-on-address style of memory, even though Vivado and Yosys both process it correctly. Normally this pattern would be used to describe a transparent (write-through) read port on a RAM, but this is meaningless for a ROM. Perhaps this would be better emitted as a more traditional ROM, something like:

```verilog
reg [31:0] mem[0:8191];
reg [31:0] mem_rdata;
always @(posedge sys_clk) begin
	mem_rdata <= mem[basesoc_rom_adr];
end

assign basesoc_rom_dat_r = mem_rdata;

initial begin
	$readmemh("mem.init", mem);
end
```

AnttiLukats, 2019-08-02:
no, Vivado does also not recognize it, I just checked. But for some reason the LUTROM on vivado takes so little resource that you do not even notice it in resource utilization summary :)

AnttiLukats, 2019-08-02:
the other way of describing ROM still yields to LUTROM on MAX10 ..

AntonKrug, 2019-08-02:
Would this be any help? http://www.gstitt.ece.ufl.edu/courses/eel4712/lectures/vhdl/qts_qii51007.pdf

AnttiLukats, 2019-08-02:
HA HA HA, sorry folks - I guess the issue is solved, I guess will test - in 10M08 there are not enough Block RAM's so quartus decided silently that well I do it LUTROM way.. without super bold warning about it, I guess this is it.

AnttiLukats, 2019-08-02:
no, it was double or triple errors, that blanked the correct solution:

the code AS GENERATED for sure does not infer ROM, so more standard rom inferred template should be used, then if the MAX10 configuration mode is correct AND there are enough RAMs then real ROM is inferred. So the LiteX does generate bad verilog, currently only solution is to open the generated quartus project, PATCH each time manually the ROM section and recompile.. :(

enjoy-digital, 2019-08-02:
@AnttiLukats we'll fix don't worry :), that's something i already saw in the past, but i probably haven't noticed it recently since i was small design on medium size FPGA. It's indeed related to the memories the ROM, but also probably the SRAM and L2 cache.

AnttiLukats, 2019-08-02:
no, only the ROM is affected, all other memories infer correctly.

enjoy-digital, 2019-08-02:
Can you share how you are patching the verilog to make it work?

AnttiLukats, 2019-08-02:
reg [31:0] mem[0:4095];
reg [31:0] mem_rdata;

always @(posedge sys_clk) begin
	mem_rdata <= mem[rom_adr];
end

assign rom_dat_r = mem_rdata;

initial begin
	$readmemh("mem.init", mem);
end

//reg [31:0] mem[0:4095];
//reg [11:0] memadr;
//always @(posedge sys_clk) begin
//	memadr <= rom_adr;
//end

//assign rom_dat_r = mem[memadr];
//
//initial begin
//	$readmemh("mem.init", mem);
//end

enjoy-digital, 2019-08-05:
Fixed with https://github.com/enjoy-digital/litex/commit/a9fe2788a2308aa59d13cb505d4ad7e10797f45a.

closed 2019-08-05 by enjoy-digital