enjoy-digital/litex #229: Module `\FD1S3BX' is not part of the design when using Trellis

DurandA, 2019-08-02:
I tried to build a SoC for the _ECP5 Evaluation Board_ using `python -m litex_boards.official.targets.versa_ecp5 --gateware-toolchain=trellis`.

Unfortunately, the following error is triggered:
```
ERROR: Module `\FD1S3BX' referenced in module `\top' in cell `\FD1S3BX_3' is not part of the design.
Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/duranda/devel/versa-litex/venv/lib/python3.6/site-packages/litex_boards/official/targets/versa_ecp5.py", line 148, in <module>
    main()
  File "/home/duranda/devel/versa-litex/venv/lib/python3.6/site-packages/litex_boards/official/targets/versa_ecp5.py", line 145, in main
    builder.build()
  File "/home/duranda/devel/versa-litex/venv/lib/python3.6/site-packages/litex/soc/integration/builder.py", line 192, in build
    toolchain_path=toolchain_path, **kwargs)
  File "/home/duranda/devel/versa-litex/venv/lib/python3.6/site-packages/litex/soc/integration/soc_core.py", line 494, in build
    return self.platform.build(self, *args, **kwargs)
  File "/home/duranda/devel/versa-litex/venv/lib/python3.6/site-packages/litex/build/lattice/platform.py", line 33, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/duranda/devel/versa-litex/venv/lib/python3.6/site-packages/litex/build/lattice/trellis.py", line 194, in build
    _run_script(script)
  File "/home/duranda/devel/versa-litex/venv/lib/python3.6/site-packages/litex/build/lattice/trellis.py", line 107, in _run_script
    raise OSError("Subprocess failed")
OSError: Subprocess failed
```

Both `litex` and `litex-boards` were installed from latest master.

> According to [_FPGA Libraries Reference Guide_](https://www.latticesemi.com/-/media/LatticeSemi/Documents/UserManuals/EI/FPGALibrariesReferenceGuide34.ashx?document_id=50882), The _FD1S3BX_ is a Positive Edge Triggered D Flip-Flop with Positive Level Asynchronous Preset.

Is this error something related to LiteX or does Trellis lack some features yet?

daveshah1, 2019-08-02:
Is your Yosys up to date? This should be supported.

DurandA, 2019-08-02:
Updating Yosys solved my issue, thanks!

My Yosys was slightly outdated as it was built in January. I didn't expect Yosys to handle architecture-specific blocks.

closed 2019-08-02 by DurandA