enjoy-digital/litex #230: Top-level input 'D0' also driven by D0.O on versa_ecp5

DurandA, 2019-08-02:
When building a soc with `python -m litex_boards.official.targets.versa_ecp5 --gateware-toolchain=trellis` with recent builds of _Yosys_ and _NextPNR_, I get the following error:
```
Info: Importing module CCU2C
Info: Rule checker, verifying imported design
Info: Importing module L6MUX21
ERROR: Top-level input 'D0' also driven by D0.O.
ERROR: Loading design failed.
0 warnings, 2 errors
Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/duranda/devel/versa-litex/venv/lib/python3.6/site-packages/litex_boards/official/targets/versa_ecp5.py", line 148, in <module>
    main()
  File "/home/duranda/devel/versa-litex/venv/lib/python3.6/site-packages/litex_boards/official/targets/versa_ecp5.py", line 145, in main
    builder.build()
  File "/home/duranda/devel/versa-litex/venv/lib/python3.6/site-packages/litex/soc/integration/builder.py", line 192, in build
    toolchain_path=toolchain_path, **kwargs)
  File "/home/duranda/devel/versa-litex/venv/lib/python3.6/site-packages/litex/soc/integration/soc_core.py", line 494, in build
    return self.platform.build(self, *args, **kwargs)
  File "/home/duranda/devel/versa-litex/venv/lib/python3.6/site-packages/litex/build/lattice/platform.py", line 33, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/duranda/devel/versa-litex/venv/lib/python3.6/site-packages/litex/build/lattice/trellis.py", line 194, in build
    _run_script(script)
  File "/home/duranda/devel/versa-litex/venv/lib/python3.6/site-packages/litex/build/lattice/trellis.py", line 107, in _run_script
    raise OSError("Subprocess failed")
OSError: Subprocess failed
```

My first guess is that there is a signal conflict in [_versa_ecp5_ target](https://github.com/litex-hub/litex-boards/blob/master/litex_boards/official/targets/versa_ecp5.py). I also don't see any reference to `D0` in [_versa_ecp5_ platform](https://github.com/litex-hub/litex-boards/blob/master/litex_boards/official/platforms/versa_ecp5.py). Is there a missing constraint?

daveshah1, 2019-08-02:
Now that you've updated Yosys, you need to update nextpnr too.

DurandA, 2019-08-02:
> Now that you've updated Yosys, you need to update nextpnr too.

Everything was freshly (re)installed!

DurandA, 2019-08-02:
My bad, sorry for wasting your time again. This was fixed very recently in _NextPNR_ as my build was ~30 days old.

Many thanks!

closed 2019-08-02 by DurandA