enjoy-digital/litex #231: System UART using FTDI FT245 FIFO mode

AnttiLukats, 2019-08-02:
for some boards it would be cool if the regular SoC UART perpiheral could use FT2232 fifo mode FT245 Async, as example this board

https://wiki.trenz-electronic.de/display/PD/TEI0015+TRM

has channel B in FIFO mode by default, using nios avalon FIFO to FT245 ip. 
Right now porting litex to those boards may require changing FTDI EEPROM configuration all the time..

enjoy-digital, 2019-08-03:
It's already possible to disable the default UART (with_uart=False on SoCCore/SoCSDRAM) and instantiate your custom UART. (the submodule needs to be called uart and have similar registers than the default UART). FT245PHYAsynchronous can be reused for that (https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/usb_fifo.py#L145) and specific software will probably be needed on the Host. So that's something that can already be done and that could be integrated in the target file since a bit specific to this board.

AnttiLukats, 2019-08-03:
hrm, what would be VERY cool would be UART like component, than can be use AS IF it would be uart from the LiteX SoC but with the rx and tx path exported as simple 8 bit wide streams - then it would be used choice what he connects there.

As of HOST requirements the beauty with async FIFO mode is that the PC host does not notice that its not real UART and you can get up to 10MB/s from to the LiteX using regular serial port drivers on the host, we are using this to stream ADC data from ADAQ7988 over UART using Audio-UART protocol and Sodira ADC

http://www.dsp4swls.de/sodirasdr/sodirasdreng.html

currently the ADC and FT2232 is handled by the free NIOS/e what is very slow, so there are lots of tricks and CUSTOM instruction to get NIOS to spit out data to FT245 like FIFO fast enough, custom instruction + asm coding and partial unrolling involved.

it would be interesting to compare if VexRiscv can handle it without dirty tricks ;)

enjoy-digital, 2019-08-03:
That's indeed interesting it's assimilated as a real UART by the Host. All the element are existing to make it work, i'll probably give it a try.

AnttiLukats, 2019-08-04:
please if you can that would really look like UART then on both ends of the wire while giving a lots of bandwidth boost, all transparently and without any change to code on both ends.

enjoy-digital, 2019-08-04:
This should now be possible with https://github.com/enjoy-digital/litex/commit/ce5c58592b95d9115979de8e6a2069dc705ffd31. 

If you want to test it, you can define the FT245 IOs for your board like this:
https://github.com/enjoy-digital/litex/blob/master/litex/boards/platforms/minispartan6.py#L66-L74

And then in SoCCore or SoCSDRAM, just change uart_name to your new resource name (usb_fifo in this case): https://github.com/enjoy-digital/litex/blob/master/litex/soc/integration/soc_core.py#L164

enjoy-digital, 2019-08-05:
I just tested it on the minispartan6 by just changing uart_name from  `serial` , to `usb_fifo` and it's working fine :) I wasn't aware FT245 FIFO mode could be as a regular UART, thanks for sharing.

closed 2019-08-05 by enjoy-digital

AnttiLukats, 2019-08-05:
cool, I will try out also, there is one more trick - some boards have BDBUS0..3 wired, they can be used in high speed opto mode, this gives so maybe 2MB/s data transfer rate what is also nice already, and the opto serial mode RTL is even smaller than UART code ;), we use it in several cases, with interface clock always running and small fifo on rx side, works like charm