enjoy-digital/litex #232: Supporting ECP5 Evaluation Board

DurandA, 2019-08-02:
I am in the process of adding support to the ECP5 Evaluation Board to LiteX. I am basically taking the `versa_ecp5` platform and removing the DDR/Ethernet components.

I have a few questions regarding this process:
1. How do I find the correct [PCF definition](https://github.com/litex-hub/litex-boards/blob/bbf0e770e9a1e047e5cc34325d235d9fd1e68f6a/litex_boards/official/platforms/versa_ecp5.py#L218-L250) for my board? Was it initially generated by the Diamond software?
2. Is there any other purpose of defining _IOStandard_ than preventing connection of incompatible signals? I expected all the I/O pins to have the same _IOStandard_ (except the SERDES pins). However [it doesn't seem to be the case](https://github.com/litex-hub/litex-boards/blob/bbf0e770e9a1e047e5cc34325d235d9fd1e68f6a/litex_boards/official/platforms/versa_ecp5.py#L13-L31). I guess that the only way to correctly define it is by careful read of the [ECP5 and ECP5-5G Family Data Sheet](https://www.latticesemi.com/view_document?document_id=50461).
3. What is the purpose of [cd_init](https://github.com/litex-hub/litex-boards/blob/bbf0e770e9a1e047e5cc34325d235d9fd1e68f6a/litex_boards/official/targets/versa_ecp5.py#L28) and [cd_por](https://github.com/litex-hub/litex-boards/blob/bbf0e770e9a1e047e5cc34325d235d9fd1e68f6a/litex_boards/official/targets/versa_ecp5.py#L29) clock domains?
4. Why does the SoC [generate a `cd_sys2x.clk` 200 Mhz clock from the 100 Mhz clock](https://github.com/litex-hub/litex-boards/blob/bbf0e770e9a1e047e5cc34325d235d9fd1e68f6a/litex_boards/official/targets/versa_ecp5.py#L56-L74) before deriving the `cd_sys.clk`? Is it for better precision when the system clock frequency needs to be rounded?

I will do a pull request to [litex-boards](https://github.com/litex-hub/litex-boards) to include it in community boards when I will be done with it.

enjoy-digital, 2019-08-02:
Hi @DurandA,

here are the answers to your questions:
1) It was initially generated by Diamond, but we now have openocd support for loading bistreams, as done in https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/make.py#L138, maybe we should only support for versa_ecp5 now, at least that's the prefered way for your board.
2) Each of the FPGA bank can use a different VCC rail, the IOStandard will allow the bistream to configure the IOs for the right VCC rail, you need to reuse an existing constraint file for that or as you are saying read carefully the schematic of your board.
3) cd_init is only needed for design the DDR3, it is used to the the initialization of some ECP5 primitives, cd_por is used to generate an initial power on reset, you can keep it.
4) The clocking is complicated here simply to satisfy some requirement for the DDR3 primitives, in your case you can use something more simple like for the ULX3S: https://github.com/litex-hub/litex-boards/blob/master/litex_boards/partner/targets/ulx3s.py#L38-L44

> I will do a pull request to litex-boards to include it in community boards when I will be done with it.

Thanks :)

DurandA, 2019-08-05:
I got a basic version working on the [ecp5-evn branch](https://github.com/DurandA/litex-boards/tree/ecp5-evn).

However, a few things are missing/not working:
* Serial using the built-in FTDI controller does not work (requires 0 Ω resistors on R34 and R35). I am not familiar with the FT2232HL as it seems to do much more than USB<->serial.
* Only the U1 clock (12 Mhz) was tested. X5 oscillator (50 Mhz) is not populated by default—I ordered one. The X2 (200 Mhz) clock [cannot be used as a normal clock source](https://github.com/SymbiFlow/prjtrellis/issues/85#issuecomment-518216381). I still have to test the PLL with U1 and X5.
* Most external connector pins are not defined yet
* Using `litex.build.openocd.OpenOCD` as a programmer does not work

Concerning OpenOCD integration, I am currently flashing the ECP5 using the command `openocd -f prog/ecp5-evn.cfg -c "transport select jtag; init; svf soc_basesoc_ecp5eval/gateware/top.svf; exit"`. I tried to use `litex.build.openocd.OpenOCD` [in `create_programmer()`](https://github.com/DurandA/litex-boards/blob/4753b28a1512caa5b6559ac8f8916da8f9bc02bc/litex_boards/community/platforms/ecp5_evn.py#L134-L137) and then using the flash function. However, I am not sure if [this command sequence](https://github.com/enjoy-digital/litex/blob/236070fdcf9d9ef3fec45f046b312e278f2ebe11/litex/build/openocd.py#L28-L32) also works for this device (I will try). Anyway, `find_flash_proxy()` always crash with `prog.flash(0, "soc_basesoc_ecp5eval/gateware/top.svf")`:
```
  File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/duranda/devel/versa-litex/litex-boards/litex_boards/community/targets/ecp5_evn.py", line 79, in <module>
    main()
  File "/home/duranda/devel/versa-litex/litex-boards/litex_boards/community/targets/ecp5_evn.py", line 76, in main
    prog.flash(0, "soc_basesoc_ecp5eval/gateware/top.svf")
  File "/home/duranda/devel/versa-litex/venv/lib/python3.6/site-packages/litex/build/openocd.py", line 26, in flash
    flash_proxy = self.find_flash_proxy()
  File "/home/duranda/devel/versa-litex/venv/lib/python3.6/site-packages/litex/build/generic_programmer.py", line 24, in find_flash_proxy
    fullname = tools.cygpath(os.path.join(fulldir, self.flash_proxy_basename))
  File "/home/duranda/devel/versa-litex/venv/lib/python3.6/posixpath.py", line 94, in join
    genericpath._check_arg_types('join', a, *p)
  File "/home/duranda/devel/versa-litex/venv/lib/python3.6/genericpath.py", line 149, in _check_arg_types
    (funcname, s.__class__.__name__)) from None
TypeError: join() argument must be str or bytes, not 'NoneType'
```

Should I inherit `OpenOCD` and override `flash()` in [platform.ecp5_evn.Platform](https://github.com/DurandA/litex-boards/blob/4753b28a1512caa5b6559ac8f8916da8f9bc02bc/litex_boards/community/platforms/ecp5_evn.py#L119) or do you have a preferred solution?

enjoy-digital, 2019-08-06:
Hi @DurandA,

to merge it, i think it would be good to be able to run a minimal SoC on it. Can you mount the two 0 ohms resistors to get the UART, and then test a simple SoC with it? First with a 12MHz system clock, then using the PLL in increasing sys_clk to a higher frequency. For the OpenOCD integration, no need to worry about that for now, i need to integrate that correctly for the Versa too, and could apply that to the ECP5 EVN after.

If you want to ease collaboration, can you create a pull request with the ecp5_evn on litex-boards, i'll help you creating the minimal SoC.

DurandA, 2019-08-06:
Hi @enjoy-digital,

> to merge it, i think it would be good to be able to run a minimal SoC on it.

What do you mean by a simple SoC? E.g., `python -m litex_boards.community.targets.ecp5_evn --cpu-type vexriscv --cpu-variant=lite` runs fine as I can see the BIOS prompt. Other targets from _litex-boards_ don't do much more but I can add some sensible defaults for the board.

> Can you mount the two 0 ohms resistors to get the UART, and then test a simple SoC with it?

Sorry, my explanation was not crystal clear. I already soldered this resistors but I am unable to access the serial through the FT2232HL for whatever reason. I currently use some I/O pins connected to an external UART adapter.

> First with a 12MHz system clock, then using the PLL in increasing sys_clk to a higher frequency.

I will do it at the same time I try the 50 Mhz optional clock. I should receive the oscillator tomorrow.

Thanks!

enjoy-digital, 2019-08-06:
Ah ok, that's fine if you are already able to run a simple SoC. I thought you were not able to get the UART so was not able to test. If you want you can already to a pull request to litex-boards.

DurandA, 2019-08-08:
I was able to clear up the remaining issues:

> * Serial using the built-in FTDI controller does not work (requires 0 Ω resistors on R34 and R35). I am not familiar with the FT2232HL as it seems to do much more than USB<->serial.

This is fixed by configuring the FT2232H with [UART virtual COM on port B](https://numato.com/kb/configuring-ft2232h-usb-serial-converter-saturn-spartan-6-module/). The configuration is stored on the external EEPROM so it is permanent.

> * Only the U1 clock (12 Mhz) was tested. X5 oscillator (50 Mhz) is not populated by default—I ordered one. The X2 (200 Mhz) clock [cannot be used as a normal clock source](https://github.com/SymbiFlow/prjtrellis/issues/85#issuecomment-518216381). I still have to test the PLL with U1 and X5.

It works fine using an optional oscillator on X5. The enable pin of the oscillator is connected directly to the FPGA so it needs to be driven. I defined the target so so the user can optionally add the `--use-x5-clock` switch.

> * Most external connector pins are not defined yet

Due to the very large amount of breakout headers on the board, I will restrict the definition to the most common ones.

I will create a pull request in the next few days after some cleanup.

enjoy-digital, 2019-08-09:
Great, thanks.

closed 2019-08-09 by enjoy-digital