enjoy-digital/litex #233: Load bios from SPI

danielkucera, 2019-08-03:
With misoc it is possible to set integrated rom size to zero and load bios from SPI like this:
https://github.com/danielkucera/flipsyfat/blob/sdstream/flipsyfat/targets/qm_xc6slx16_sdram.py#L162
I havent found an example for litex. Is this possible? How to do this? My bios doesn't fit into an integrated rom of maximum size :/

mithro, 2019-08-03:
The MimasV2 platform in LiteX-BuildEnv uses the BIOS from spiflash, see https://github.com/timvideos/litex-buildenv/blob/master/targets/mimasv2/base.py

AnttiLukats, 2019-08-03:
I guess there are two uses cases, one is bootstrap from SPI flash to some on-chip or off-chip RAM, the other is running BIOS from SPI in XiP mode

enjoy-digital, 2019-08-06:
Linux-on-LiteX-Vexriscv project also show how to instantiate SPIFlash module: https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/soc_linux.py#L42
To boot from it, user then just to set cpu_reset_address to the location the bios is stored in the SPIFlash. (https://github.com/enjoy-digital/litex/blob/master/litex/soc/integration/soc_core.py#L150)
I hope the question is answered.

closed 2019-08-06 by enjoy-digital

danielkucera, 2019-08-11:
I have instantiated SPI flash and I can read from it with integrated bios and I see correct code of bios by running:
```
mr 0x20080000 100
```
Now I've set 
```
            kwargs['integrated_rom_size']=None
            kwargs['cpu_reset_address']=self.mem_map["spiflash"]+platform.gateware_size
```
But I am getting:
```
migen.fhdl.module.FinalizeError: CPU needs a rom to be registered with SoC.register_mem()
```
And I don't exactly understand what to do because I have registered mem:
```
        self.register_mem("spiflash", self.mem_map["spiflash"],
                            self.spiflash.bus, size=platform.spiflash_total_size)
```

danielkucera, 2019-08-11:
Ok, solved! I had to add:
```
        bios_size = 0xa000
        self.add_memory_region("rom", kwargs['cpu_reset_address'], bios_size)
```

enjoy-digital, 2019-08-12:
@danielkucera: good, i'll make sure there is an example on how to use it.