enjoy-digital/litex #237: avalanche target file seems to be missing?

AnttiLukats, 2019-08-10:
there is platform file, but no target also no target for any other microsemi board?

enjoy-digital, 2019-08-10:
The avalanche platform can be used with the simple target (https://github.com/enjoy-digital/litex/blob/master/litex/boards/targets/simple.py):
`litex_simple litex.boards.platforms.avalanche `

To use the DDR3, the PHY from Microsemi is still used, so a target with DDR3 is not integrated the target directly in LiteX but can be found here:
https://github.com/antmicro/avalanche
and instructions to use it here:
https://risc-v-getting-started-guide.readthedocs.io/en/latest/linux-avalanche.html

closed 2019-08-10 by enjoy-digital

AnttiLukats, 2019-08-10:
i see, did not find that repo somehow