enjoy-digital/litex #239: Quartus: be sure timing constraints are applied correctly

enjoy-digital, 2019-08-11:
See https://github.com/litex-hub/litex-boards/issues/9:
> The problem was in SDC file generated it does try to timing constraints to PORTS while using internal clock net names, this causes those constraints to be ignored by Quartus.

AnttiLukats, 2019-08-12:
not sure what the better fix is, to force the constraint to use net, or then set the constraint on actual input port, both would work..

a function that would "emit" a string to SDC file from target python could also do the trick

enjoy-digital, 2019-08-12:
I did some tests this morning using net, quartus was removing the net since it was equivalent to the port (even when setting a keep attribute on the wire). I'll need to spend more time looking at that, for now, we can constraint using the port's name with `lookup_request`:

````python
        platform.add_period_constraint(platform.lookup_request("eth_clocks").tx, 1e9/12.5e6)
        platform.add_period_constraint(platform.lookup_request("eth_clocks").rx, 1e9/12.5e6)
        platform.add_false_path_constraints(
            platform.lookup_request("clk12"),
            platform.lookup_request("eth_clocks").tx,
            platform.lookup_request("eth_clocks").rx
        )

AnttiLukats, 2019-08-12:
that will do also! tnx

enjoy-digital, 2019-09-10:
Cyclone 10 Ref Kit can now be used as reference to apply constraints on Intel design: https://github.com/litex-hub/litex-boards/blob/master/litex_boards/partner/targets/c10lprefkit.py#L130-L136

closed 2019-09-10 by enjoy-digital