enjoy-digital/litex #242: adding signaltap

AnttiLukats, 2019-08-14:
https://github.com/timvideos/litex-buildenv/wiki/LiteX-for-Hardware-Engineers

there is explained how to add chipscope, but trying todo similar thing with quartus, add the 

qsys ip core file as verilog source, and that does not work out well

enjoy-digital, 2019-08-14:
Sorry, i'm not sure to understand the issue? You want to specific command to add qsys files?

AnttiLukats, 2019-08-15:
timvideos pages explains how to add xilinx chipscope XCI file using
        self.platform.add_source("ila.xci")
when trying similar thing for Quartus
        self.platform.add_source("ila.qsys")
then the file is added, but also decorated as systemverilog file

set_global_assignment -name SYSTEMVERILOG_FILE X:/GIT/litex-boards/litex_boards/partner/targets/ila.qsys -library work

so quartus does not recognize it as plaform builder ip file, and does not compile unless qsys file is removed and added back to the project

similar command for xilinx add the xci file as generic file and this is accepted by xilinx toolchain

so basically it is not about signaltap or quartus, more generic how to add files to the project without them automatically tagges as verilog files, they can be any files

enjoy-digital, 2019-08-15:
Implemented with https://github.com/enjoy-digital/litex/commit/6e6fe83af3a4acbc0d21043656813ea25f3a2f01, you can now use `platform.add_ip` for that. (that's also the prefered way to do it with Xilinx tools for .xci)

closed 2019-08-15 by enjoy-digital