enjoy-digital/litex #249: integer parameter for instance

AnttiLukats, 2019-08-26:
how to define a instance parameter that is simple integer? not string, not vector, not float ?

module.specials += Instance("alt_jtag_atlantic",
    p_INSTANCE_ID = 0,
    p_LOG2_RXFIFO_DEPTH =0, 

both 0 convert to 1'd0, no strange that quartus accepts 1'd0 for the first parameter, but not for the second one, it has to be 0, not 0.0 or 1'd0 !

it seems workaround this time is to use "0", this is for some reason eaten by quartus, but I would still like to know how to define integer instance param, assuming this is supported at all

enjoy-digital, 2019-08-29:
All integers are emitted this way for Migen instances, this is done here: https://github.com/m-labs/migen/blob/master/migen/fhdl/specials.py#L205. Changing that to str(p.value.value) produce an integer, but according to me, the fact that Quartus does not accept the 1'd notation is a bug, so not sure we do the change just for that. The best is probably to workaround the Quartus bug as you already did, just for this parameter.

closed 2019-08-29 by enjoy-digital

AnttiLukats, 2019-08-29:
there may be MORE "bugs" like this, possible so having an escape to emit REAL integers as p_xxx would be good for long term, but for now the workaround is - there is actually 2 different bugs:

https://github.com/micro-FPGA/litex-boards/blob/70ed1d79a8df28a36cc4cc09d559416ba2c6c53d/litex_boards/partner/targets/atlantic.py#L32

first parameter has to be OMMITED as any output from migen would yield to error, both 1b'0 and "0"
2,3 parameter has to be set as string (can not be ommited)

for me this is ok workaround, as long as there will be other similar cases with no workarounds..

enjoy-digital, 2019-08-29:
Does it work correctly with the Migen change i suggested? If so, maybe we should do the change in Migen, but we also need to be sure it's not introducing others issues.

AnttiLukats, 2019-09-19:
ups, I did not realize that it was my task to actually patch and test the migen change, so I havent done it. There workarounds so far work, so there is no urgent need, but it may come as showstopper with some other module later.