enjoy-digital/litex #250: creating JTAG UART module

AnttiLukats, 2019-08-27:
https://github.com/micro-FPGA/litex-boards/blob/master/litex_boards/partner/targets/atlantic.py

something seems to be wrong, the CSR's are not generated but I am failing to see why ..

the idea is to say "with_uart=False" and later add the Quartus/Altera JTAG atlantic UART that would be recognized and used by the BIOS.. several Altera boards do not have real UART on board, so jtag uart would be nicel to have feature, the "nios_terminal" can then be used to talk to those boards over main usb-jtag port

mithro, 2019-08-27:
IIRC @jordens has done something like this before...

mithro, 2019-08-27:
Looks like I have something like this here -> https://github.com/timvideos/HDMI2USB-litex-firmware/compare/master...mithro:jtag-uart

enjoy-digital, 2019-08-27:
It ould it be that you are using:
````
my_module = MyModule()
self.submodules += my_module
````
instead of:
````
self.submodules.my_module = MyModule()
````
which is required for modules with CSRs.

enjoy-digital, 2019-08-27:
@mithro, @AnttiLukats: that would indeed be very nice to have UART over JTAG, even for boards that have UART since this would allow a using the same cable for programming/UART.

AnttiLukats, 2019-08-29:
@enjoy-digital I got the atlantic_uart working on Intel C10LP kit (terasic with byteblaster on-board) - I am not sure how much "cleaning" my code would need to be merged into main litex - in any case it known working and tested solution. The fifo and clock domain crossing are in altera megafunction, fifo size fix encoded to 32 bytes, yields pretty good performance (with smaller fifos long prints are slow..) there are many terasic-intel boards with no "uart" so this litex module would be really useful, I have at least 3 boards that I could not use, as there was no uart..

@mithro @enjoy-digital indeed, second "jtag-uart" would allow debug separation, if debug and user ip are both on uart channel.

@mithro your code is pretty much minimal, there are a few more similar projects I am aware of and I have myself done something similar also, years ago

=> I think the "jtag_uart" should be considered as part of larger jtag-debug-instrumentation framework, so it may be too early to add some uart only implementation now, so I close the issue here. I will do some more homework on 1149.7 and 1687 SiB etc to see what if any of that can be re-used to implement open source jtag-hub infra.

closed 2019-08-29 by AnttiLukats