enjoy-digital/litex #258: Quartus CLOCK constraint on generated clock (signal)

AnttiLukats, 2019-09-14:
There seems to be no way to force timing constraint for internal signal (generated clock) - this is sometimes REALLY needed, and it seems there is no feasible workaround at the moment.

optionally a function to "insert" arbitrary text to generated SDC file would also help, and might be a good idea to have

enjoy-digital, 2019-09-14:
Can you explain how it can be done in a traditional design to see what need to be done?

AnttiLukats, 2019-09-15:
in SDC file
create_clock -name {fiftyfivenm_oscillator~OBSERVABLEOSCENAREG} -period 8.600
litex creates only clocks for ports
create_clock -name clk12 -period 83.0 [get_ports {clk12}]

so if it could be possible to emit timing constraint without the [get_port {}] then it would be sufficient.

but for most flexibility to "emit arbitrary text to SDC" could also be implemented

enjoy-digital, 2019-12-06:
With https://github.com/enjoy-digital/litex/commit/4c9af635d21ff24e2bc7433e3b7ae510d5198b21 it's now possible to apply timing constraints on internal nets. It's also possible to add additional sdf or qsf command from the design like this:
```
platform.toolchain.sdc_additional_commands.append("create_clock ...")
platform.toolchain.sdc_additional_commands.append("set_false_path ...")
```

closed 2019-12-06 by enjoy-digital