enjoy-digital/litex #265: Add module-level documentation

xobs, 2019-09-23:
Continuing from https://github.com/enjoy-digital/litex/pull/259#issuecomment-533648773

I've come up with a solution for adding module-level documentation.  It is very similar to `AutoCSR`, in that an auto-documented module would inherit from the `AutoDocument` class.  This would cause the document generation engine to gather all sub-objects that inherit from the `ModuleDocumentation` class.

Here's an example of what it could look like: https://doctest.xobs.io/usb.html

This particular document was generated from something like this:

```python
class TriEndpointInterface(Module, AutoCSR, AutoDocument):
    def __init__(self, iobuf, debug=False):
        self.background = ModuleDocumentation(title="USB Device Tri-FIFO", description="""
            This is a three-FIFO USB device.  It presents one FIFO each for ``IN``, ``OUT``, and
            ``SETUP`` data.  This allows for up to 16 ``IN`` and 16 ``OUT`` endpoints
            without sacrificing many USB resources.""")
            self.submodules.debug_bridge = ClockDomainsRenamer("usb_12")(debug_bridge)

class USBWishboneBridge(Module, AutoDocument):
    def __init__(self, usb_core, clk_freq=12000000, magic_packet=0x43):
        self.wishbone = wishbone.Interface()
        self.background = ModuleDocumentation(title="USB Wishbone Bridge", description="""
            This bridge provides a transparent bridge to the target device's Wishbone bus over USB.
            It can operate without interfering with the device's USB stack.  It is simple enough to
            be able to work even if the USB stack is not enumerated, though the host may not cooperate.""")

        self.protocol = ModuleDocumentation(title="USB Wishbone Debug Protocol", description="""
        The protocol transfers four bytes a time in big-endian (i.e. USB) order.  It uses SETUP packets
        with the special type (0x43) as an `attention` word.  This is then followed by an ``OUT`` packet.

            .. wavedrom::
                :caption: Write to Wishbone

                { signal: [
                    ['Request',
                        {  name: 'data',        wave: 'x222...22x', data: '0x43 0x00 [ADDRESS] 0x04 0x00'   },
                        {  name: 'data bits',   wave: 'xxx2222xxx', data: '7:0 15:8 23:16 31:24'},
                        {  name: 'usb meaning', wave: 'x222.2.2.x', data: 'bReq bTyp wValue wIndex wLength' },
                        {  name: 'usb byte',    wave: 'x22222222x', data: '1 2 3 4 5 6 7 8'                 }
                    ],
                    {},
                    ['Payload',
                        {  name: 'data',        wave: 'x3...x', data: '[DATA]'},
                        {  name: 'data bits',   wave: 'x3333x', data: '7:0 15:8 23:16 31:24'},
                        {  name: 'usb meaning', wave: 'x3...x', data: 'OUT'  },
                        {  name: 'usb byte',    wave: 'x3333x', data: '1 2 3 4'}
                    ]
                ]}
...
```

This approach has several nice properties:

* sub-documentation gets included (e.g. the `USBWishboneBridge` is a sub-module of `USB`
* The way `ModuleDocumentation` is implemented, one can simply subclass it and fill the module's documentation into the docstring.  This achieves @enjoy-digital 's goal of keeping documentation in the docstring
* It is also possible to refer to an external file, which sounds like what @bunnie was looking for
* It allows for multiple sections within a single documentation block
* It doesn't require any extra modules when generating the source documentation file

enjoy-digital, 2019-09-23:
The result is really nice (and thanks for the special effort allowing keeping the documentation in the  docstring) I would maybe use shorten names:  AutoDoc, ModuleDoc, but otherwise that seems really fine.

xobs, 2019-09-23:
If I was to make a pull request, where would it go?  The current `ModuleDocumentation` module is located at https://github.com/xobs/lxsocdoc/blob/c190ed5fba131d24b18284060bb63c53a88f5c04/lxsocdoc/module.py

enjoy-digital, 2019-09-23:
What about `litex.soc.integration.doc` ?

enjoy-digital, 2019-09-23:
Great, thanks for your work.

enjoy-digital, 2019-10-06:
Added with https://github.com/enjoy-digital/litex/pull/266.

closed 2019-10-06 by enjoy-digital