enjoy-digital/litex #273: Quick start guide (for advanced users) - RISC-V toolchain not detected

pcotret, 2019-10-08:
Following the quick start for advanced users, the RISC-V toolchain is not detected as I get the following error:
OSError: Unable to find any of the cross compilation toolchains:
- riscv64-unknown-elf
- riscv32-unknown-elf
- riscv-none-embed

In other words, `litex/litex/soc/integration/export.py` returns `None` when it tries `which("riscv64-unknown-elf-gcc")`

Strange thing, the RISC-V toolchain is in my PATH...
`pascal@laptop:$ riscv64-unknown-elf-gcc
riscv64-unknown-elf-gcc: fatal error: no input files`

I'll investigate this issue a bit later.

enjoy-digital, 2019-10-08:
Thanks for reporting, are you using a specific OS?

Do you have the same results if you run the following commands in the python3 shell?:
````
>>> from shutil import which
>>> which("riscv64-unknown-elf-gcc")
'/opt/riscv/bin/riscv64-unknown-elf-gcc'
>>> 
````

pcotret, 2019-10-09:
Yes, that's the thing I've tried afterwards :/

pcotret, 2019-10-09:
Yes, that's the thing I've tried afterwards... :/

pcotret, 2019-10-09:
I've found my mistake:
- Running `litex_setup.py` as a normal user. Doesn't work as it it want to write stuff is `usr/local/lib/python`.
- Running `litex_setup.py` as sudo.
- Running the next steps as written in the `readme`.

Things would have worked better if I ran `./litex_setup.py init install --user` (my bad...)

In my current setup, I had to `chown` my `enjoy-digital` directory (where `litex_setup.py` was downloaded) to myself before it works :100:

closed 2019-10-09 by pcotret