enjoy-digital/litex #281: ICAPBitstream fails to load long bitstreams

mateusz-holenko, 2019-10-16:
The version of ICAP (now renamed to ICAPBitstream) we PRed in #269 was capable of handling long bitstreams. The simplifications introduced in 41ad08e8eff65a44803beb9259a0d44b29950dab unfortunately caused it to break. We prepared [some fixes](https://github.com/antmicro/litex/commit/7866d879274f2299b200e1b886350e7e6209546c), but they seem not to address all the problems.

What's more the simplifications changed CSRs thus breaking compatibility with [our linux driver](https://github.com/antmicro/linux-on-litex-vexriscv/commit/720b39f462dd0362145fc47599157acfdc170e0a) we were planning to publish to [linux-on-litex-vexirscv](https://github.com/litex-hub/linux-on-litex-vexriscv) repository.

The question is: are those simplifications really needed?  Of course if there are any problems with our implementation we are happy to discuss it, but it would be great if we could keep the API (CSRs).

enjoy-digital, 2019-10-16:
Sorry if the simplification are causing trouble, but we should try to have the minimal code to ease maintenance in the future. I just reviewed the simplication i made and did another simplification pass with https://github.com/enjoy-digital/litex/commit/97a77b950c436530390d43a7754cff28f86faa8c.

In your code, i understood that once the stream is started, the CPU should then be able to provide data quickly enough to ensure there is no gap on the ICAPE2 stream, was it the case? If so, the simplification i just made are going in this direction, and we just provide the user some register to push a stream of data and remove the control registers.

Does it seems fine for you? If you still have troubles, can you increase fifo_depth or icap_clk_div?

enjoy-digital, 2019-10-18:
OK great, i was just going to suggest that i could do the debug if it was too much troubles for you. Can you create a pull request? Thanks

mateusz-holenko, 2019-10-18:
Fixed by #282.

enjoy-digital, 2019-10-18:
Thanks for the pull-request. I just did a small change to always set ready to 1: https://github.com/enjoy-digital/litex/commit/4cf346a1d4d258949894f51f8b41665a1b943d5e