enjoy-digital/litex #287: Software only compiles on Windows when V=1

xobs, 2019-10-25:
Something changed, and now 

```powershell
smcro@CUBOID   yosys-bld  D:\Code\HaD\valentytest   master ≢                                                                                                                                                                                 [09:36]
❯ make -C D:\Code\HaD\valentytest\build2\software\libcompiler_rt -f D:\Code\HaD\valentytest\deps\litex\litex\soc\software\libcompiler_rt\Makefile
make: Entering directory 'D:/Code/HaD/valentytest/build2/software/libcompiler_rt'
 CC       umodsi3.o
riscv64-unknown-elf-gcc.exe: error: D:CodeHaDvalentytestdepslitexlitexsoc/software/compiler_rt/lib/builtins/umodsi3.c: No such file or directory
riscv64-unknown-elf-gcc.exe: fatal error: no input files
compilation terminated.
make: *** [D:\Code\HaD\valentytest\deps\litex\litex\soc\software\libcompiler_rt\Makefile:27: umodsi3.o] Error 1
make: Leaving directory 'D:/Code/HaD/valentytest/build2/software/libcompiler_rt'
smcro@CUBOID   yosys-bld  D:\Code\HaD\valentytest   master ≢                                                                                                                                                                                 [09:36]
❯ make -C D:\Code\HaD\valentytest\build2\software\libcompiler_rt -f D:\Code\HaD\valentytest\deps\litex\litex\soc\software\libcompiler_rt\Makefile V=1
make: Entering directory 'D:/Code/HaD/valentytest/build2/software/libcompiler_rt'
riscv64-unknown-elf-gcc -std=gnu99 -c -MD -MP -Os -march=rv32ima    -mabi=ilp32 -D__vexriscv__ -g3 -fomit-frame-pointer -Wall -fno-builtin -nostdinc -ID:\\Code\\HaD\\valentytest\\deps\\litex\\litex\\soc/software/include/base -ID:\\Code\\HaD\\valentytest\\deps\\litex\\litex\\soc/software/include -ID:\\Code\\HaD\\valentytest\\deps\\litex\\litex\\soc/common -ID:\\Code\\HaD\\valentytest\\build2\\software\\include -fexceptions -Wstrict-prototypes -Wold-style-definition -Wmissing-prototypes -D_YUGA_LITTLE_ENDIAN=1 -D_YUGA_BIG_ENDIAN=0 -Wno-missing-prototypes  D:\\Code\\HaD\\valentytest\\deps\\litex\\litex\\soc/software/compiler_rt/lib/builtins/umodsi3.c -o umodsi3.o
riscv64-unknown-elf-gcc -std=gnu99 -c -MD -MP -Os -march=rv32ima    -mabi=ilp32 -D__vexriscv__ -g3 -fomit-frame-pointer -Wall -fno-builtin -nostdinc -ID:\\Code\\HaD\\valentytest\\deps\\litex\\litex\\soc/software/include/base -ID:\\Code\\HaD\\valentytest\\deps\\litex\\litex\\soc/software/include -ID:\\Code\\HaD\\valentytest\\deps\\litex\\litex\\soc/common -ID:\\Code\\HaD\\valentytest\\build2\\software\\include -fexceptions -Wstrict-prototypes -Wold-style-definition -Wmissing-prototypes -D_YUGA_LITTLE_ENDIAN=1 -D_YUGA_BIG_ENDIAN=0 -Wno-missing-prototypes  D:\\Code\\HaD\\valentytest\\deps\\litex\\litex\\soc/software/compiler_rt/lib/builtins/udivsi3.c -o udivsi3.o
```

I'm unsure of what changed, or why it's not working.  The `&&` appears to be causing the problem now.

My current workaround is to hardcode `V=0` in `common.mk`.

mithro, 2019-10-25:
It looks like an escaping issue? `D:CodeHaDvalentytestdepslitexlitexsoc/software/compiler_rt/lib/builtins/umodsi3.c`

It is unclear to me if this is caused by https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/common.mak or the makefile in `compiler_rt @ 81fb4f0`. The fact that the `compiler_rt` value hasn't changed in 4 years makes me suspect the first one.

mithro, 2019-10-25:
Can you try reverting https://github.com/enjoy-digital/litex/commit/6d844a038a4b5244a2c459b1e7c0c4fee4baf5bf -- I can't possibly see how it is causing the behaviour you are seeing, but it is the only recent change in `common.mk`

xobs, 2019-10-25:
It definitely seems like an escaping issue.

Upon further investigation it's caused by Conda.  If I exit the Conda environment, it works just fine.

I'm unsure of how Conda is affecting things -- `make.exe` isn't getting provided by Conda.

xobs, 2019-10-25:
The culprit is `m2-bash`.  If I install this package on Conda, then the build breaks.

mithro, 2019-10-25:
The issue seems to be related to https://www.gnu.org/software/make/manual/html_node/Choosing-the-Shell.html

There is a whole complicated section on "Choosing a Shell in DOS and Windows". Reading the behaviour there, it seems like just **having** a shell in your path will change the behaviour of make. It is likely that m2-bash provides a `sh` command and hence `make` changes how it behaves?

mithro, 2019-10-25:
FYI - I have no idea what the **correct** behaviour should be here? Is it actually using `command.com` before you install `m2-bash` or some other shell?

xobs, 2019-10-25:
As a test, I created this Makefile:

```
all:
	$(foreach v,                                        \
	$(.VARIABLES), \
	$(info $(v) = $($(v))))
```

If I run this with Conda disabled, I see the following:

```
...
SHELL = sh.exe
...
```

If I run it with conda enabled and `m2-bash` installed, I get this:

```
...
SHELL = D:/Software/Miniconda3/envs/yosys-bld/Library/usr/bin/sh.exe
...
```

So the issue is `make` is picking up the ambient shell, which is doing unnecessary escaping on the command.

mithro, 2019-10-25:
@xobs In the first case, does `sh.exe` actually exist?

xobs, 2019-10-25:
In the first case, `sh.exe` does not exist.

mithro, 2019-10-25:
From reading that page above -- I think if `SHELL=sh.exe` and `sh.exe` isn't found, then make falls back to **actually** using command.com?

mithro, 2019-10-25:
Can you check what `%SHELL%` and `%COMSPEC%` values are in your environment (is that how you do environment stuff in Windows?).

xobs, 2019-10-25:
That's how you do it in `cmd.exe` (and by extension `make`), but most people use powershell nowadays, where the environment variables are all stored under the list $env:

```powershell
❯ echo $env:SHELL
❯ echo $env:COMSPEC
C:\WINDOWS\system32\cmd.exe
❯
```

So `SHELL` is undefined, and `COMSPEC` is `C:\WINDOWS\system32\cmd.exe`.  This is true both inside and outside of the conda environment.

mithro, 2019-10-25:
Can you do the equivalent of `export SHELL=$COMSPEC`?

xobs, 2019-10-25:
No change:

```powershell
❯ $env:SHELL=$env:COMSPEC
❯ echo $env:SHELL
C:\WINDOWS\system32\cmd.exe
❯ make -C D:\Code\HaD\valentytest\build\software\libcompiler_rt -f D:\Code\HaD\valentytest\deps\litex\litex\soc\software\libcompiler_rt\Makefile
make: Entering directory 'D:/Code/HaD/valentytest/build/software/libcompiler_rt'
bash.exe: warning: could not find /tmp, please create!
 CC       umodsi3.o
riscv64-unknown-elf-gcc.exe: error: D:Codehadvalentytestdepslitexlitexsoc/software/compiler_rt/lib/builtins/umodsi3.c: No such file or directory
riscv64-unknown-elf-gcc.exe: fatal error: no input files
compilation terminated.
make: *** [D:\Code\HaD\valentytest\deps\litex\litex\soc\software\libcompiler_rt\Makefile:27: umodsi3.o] Error 1
make: Leaving directory 'D:/Code/HaD/valentytest/build/software/libcompiler_rt'
❯
```

xobs, 2019-10-25:
Note that adjusting the SHELL environment variable doesn't seem to affect `make` at all:

```powershell
❯ echo $env:SHELL
C:\WINDOWS\system32\cmd.exe
❯ make |find '"SHELL"'
.SHELLFLAGS = -c
SHELL = D:/Software/Miniconda3/envs/yosys-bld/Library/usr/bin/sh.exe
.VARIABLES = GNUMAKEFLAGS <D COMPILE.m ?F CommonProgramFiles KISYS3DMOD .SHELLFLAGS MAKEFLAGS ConEmuHooks CWEAVE ?D ARFLAGS @D HOMEDRIVE @F ^D CURDIR SHELL RM CO ProgramFiles(x86) ConEmuWorkDir _NT_SYMBOL_PATH LINK.m LINK.o OUTPUT_OPTION CONDA_PREFIX_1 COMPILE.cpp PATHEXT _CONDA_EXE MAKEFILE_LIST SystemRoot TMP LINK.p ConEmuPID CUDA_PATH_V9_2 CC SystemDrive CommonProgramFiles(x86) ConEmuWorkDrive ConEmuDir CHECKOUT,v LEX.m CPP ConEmuBackHWND OPENCV_DIR LINK.cc MAKE_HOST PATH LM_LICENSE_FILE ConEmuANSI LD TEXI2DVI YACC ConEmuServerPID COMPILE.mod ConEmuDrive PROCESSOR_ARCHITECTURE LINK.r USERPROFILE LINT COMPILE.f RUBYOPT LINT.c YACC.m ALLUSERSPROFILE ComSpec NVCUDASAMPLES9_2_ROOT ConEmuDrawHWND YACC.y LOGONSERVER AR NUMBER_OF_PROCESSORS .FEATURES TANGLE NVTOOLSEXT_PATH ProgramW6432 ConEmuHWND %F CONDA_EXE COMPILE.F OS CTANGLE ConEmuAnsiLog .LIBPATTERNS COMPUTERNAME LINK.C .LOADED LINK.S PREPROCESS.r NVCUDASAMPLES_ROOT LINK.c USERNAME PUBLIC LINK.s HOME ChocolateyLastPathUpdate +D CommonProgramW6432 ProgramData ConEmuArgs CONDA_PROMPT_MODIFIER +F USERDOMAIN MAKELEVEL CONDA_DEFAULT_ENV MAKE KISYSMOD AS PREPROCESS.S ConEmuPalette COMPILE.p MAKE_VERSION FC .DEFAULT_GOAL OneDrive PREPROCESS.F %D ConEmuBuild DriverData LOCALAPPDATA CONDA_SHLVL KICAD_PTEMPLATES ANSICON_DEF WEAVE MAKE_COMMAND LINK.cpp PROCESSOR_ARCHITEW6432 F77 COMPILE.cc .VARIABLES PC *F PROCESSOR_IDENTIFIER ConEmuTask COMPILE.def LEX HOMEPATH MFLAGS *D LEX.l PSModulePath ConEmuBaseDirShort windir COMPILE.r ANSICON M2C GET SUFFIXES APPDATA TEMP _CONDA_ROOT MAKEFILES <F ConEmuCfgDir CXX COFLAGS GIT_SSH ConEmuConfig CONDA_PYTHON_EXE COMPILE.C ConEmuArgs2 ConEmuBaseDir ChocolateyToolsLocation ^F CUDA_PATH COMPILE.S LINK.F ProgramFiles COMPILE.c SESSIONNAME PROCESSOR_REVISION COMPILE.s MOZ_PLUGIN_PATH .INCLUDE_DIRS .RECIPEPREFIX MAKEINFO MAKE_TERMERR OBJC ChocolateyInstall TEX CONDA_PREFIX PROCESSOR_LEVEL F77FLAGS LINK.f LATTICE_INI_PATH USERDOMAIN_ROAMINGPROFILE
❯ make SHELL=$env:SHELL|find '"SHELL"'
.SHELLFLAGS = -c
MAKEFLAGS =  -- SHELL=C:/WINDOWS/system32/cmd.exe
SHELL = C:/WINDOWS/system32/cmd.exe
-*-command-variables-*- = SHELL=C:/WINDOWS/system32/cmd.exe
MAKEOVERRIDES = SHELL=C:/WINDOWS/system32/cmd.exe
.VARIABLES = GNUMAKEFLAGS <D COMPILE.m ?F CommonProgramFiles KISYS3DMOD .SHELLFLAGS MAKEFLAGS ConEmuHooks CWEAVE ?D ARFLAGS @D HOMEDRIVE @F ^D CURDIR SHELL RM CO ProgramFiles(x86) ConEmuWorkDir _NT_SYMBOL_PATH LINK.m LINK.o OUTPUT_OPTION CONDA_PREFIX_1 COMPILE.cpp PATHEXT _CONDA_EXE MAKEFILE_LIST SystemRoot TMP LINK.p ConEmuPID CUDA_PATH_V9_2 CC SystemDrive CommonProgramFiles(x86) ConEmuWorkDrive ConEmuDir CHECKOUT,v LEX.m CPP ConEmuBackHWND OPENCV_DIR LINK.cc MAKE_HOST PATH LM_LICENSE_FILE ConEmuANSI LD TEXI2DVI YACC ConEmuServerPID COMPILE.mod ConEmuDrive PROCESSOR_ARCHITECTURE LINK.r USERPROFILE LINT COMPILE.f RUBYOPT LINT.c YACC.m ALLUSERSPROFILE ComSpec NVCUDASAMPLES9_2_ROOT ConEmuDrawHWND YACC.y LOGONSERVER AR NUMBER_OF_PROCESSORS .FEATURES TANGLE NVTOOLSEXT_PATH ProgramW6432 ConEmuHWND %F CONDA_EXE COMPILE.F OS CTANGLE ConEmuAnsiLog .LIBPATTERNS COMPUTERNAME LINK.C .LOADED LINK.S PREPROCESS.r NVCUDASAMPLES_ROOT LINK.c USERNAME PUBLIC LINK.s HOME ChocolateyLastPathUpdate +D CommonProgramW6432 ProgramData ConEmuArgs -*-command-variables-*- CONDA_PROMPT_MODIFIER +F USERDOMAIN MAKELEVEL CONDA_DEFAULT_ENV MAKE KISYSMOD AS PREPROCESS.S MAKEOVERRIDES ConEmuPalette COMPILE.p MAKE_VERSION FC .DEFAULT_GOAL OneDrive PREPROCESS.F %D ConEmuBuild DriverData LOCALAPPDATA CONDA_SHLVL KICAD_PTEMPLATES ANSICON_DEF WEAVE MAKE_COMMAND LINK.cpp PROCESSOR_ARCHITEW6432 F77 COMPILE.cc .VARIABLES PC *F PROCESSOR_IDENTIFIER ConEmuTask COMPILE.def LEX HOMEPATH MFLAGS *D LEX.l PSModulePath ConEmuBaseDirShort windir COMPILE.r ANSICON M2C GET SUFFIXES APPDATA TEMP _CONDA_ROOT MAKEFILES <F ConEmuCfgDir CXX COFLAGS GIT_SSH ConEmuConfig CONDA_PYTHON_EXE COMPILE.C ConEmuArgs2 ConEmuBaseDir ChocolateyToolsLocation ^F CUDA_PATH COMPILE.S LINK.F ProgramFiles COMPILE.c SESSIONNAME PROCESSOR_REVISION COMPILE.s MOZ_PLUGIN_PATH .INCLUDE_DIRS .RECIPEPREFIX MAKEINFO MAKE_TERMERR OBJC ChocolateyInstall TEX CONDA_PREFIX PROCESSOR_LEVEL F77FLAGS LINK.f LATTICE_INI_PATH USERDOMAIN_ROAMINGPROFILE
❯
```

mithro, 2019-10-25:
`find` is like grep?

mithro, 2019-10-25:
Can you modify the `Makefile` to have a `SHELL=$COMSPEC` line?

xobs, 2019-10-25:
Yes, `find` is the ancient DOS equivalent of `grep`.

Setting `SHELL = C:/WINDOWS/system32/cmd.exe` inside the makefile fixes things, however.  Setting `SHELL = $(COMSPEC)` results in a build that makes me think `$(COMSPEC)` is undefined:

```powershell
❯ make -C D:\Code\HaD\valentytest\build\software\libcompiler_rt -f D:\Code\HaD\valentytest\deps\litex\litex\soc\software\libcompiler_rt\Makefile
make: Entering directory 'D:/Code/HaD/valentytest/build/software/libcompiler_rt'
process_begin: CreateProcess(NULL, -c "echo \" CC      \" umodsi3.o && riscv64-unknown-elf-gcc -std=gnu99 -c -MD -MP -Os -march=rv32ima    -mabi=ilp32 -D__vexriscv__ -g3 -fomit-frame-pointer -Wall -fno-builtin -nostdinc -ID:\\Code\\had\\valentytest\\deps\\litex\\litex\\soc/software/include/base -ID:\\Code\\had\\valentytest\\deps\\litex\\litex\\soc/software/include -ID:\\Code\\had\\valentytest\\deps\\litex\\litex\\soc/common -ID:\\Code\\had\\valentytest\\build\\software\\include -fexceptions -Wstrict-prototypes -Wold-style-definition -Wmissing-prototypes -D_YUGA_LITTLE_ENDIAN=1 -D_YUGA_BIG_ENDIAN=0 -Wno-missing-prototypes  D:\\Code\\had\\valentytest\\deps\\litex\\litex\\soc/software/compiler_rt/lib/builtins/umodsi3.c -o umodsi3.o", ...) failed.
make (e=2): The system cannot find the file specified.
make: *** [D:\Code\HaD\valentytest\deps\litex\litex\soc\software\libcompiler_rt\Makefile:27: umodsi3.o] Error 2
make: Leaving directory 'D:/Code/HaD/valentytest/build/software/libcompiler_rt'
❯
```

xobs, 2019-10-25:
Ah, of course.  Windows is case-insensitive, but Make is case-sensitive.  This works:

```
ifneq ($(ComSpec),)
SHELL = $(ComSpec)
endif
```

mithro, 2019-10-25:
Maybe we should set that?

xobs, 2019-10-25:
I guess the question is: Will this adversely impact anyone?

mithro, 2019-10-25:
@cr1901 and @enjoy-digital both use Windows, so it might be worth getting there input...

xobs, 2019-10-25:
Is `%ComSpec%` defined in their environments?  Are they using `sh.exe`?

xobs, 2019-12-23:
We ran into this issue in the Fomu workshop.  Someone showed up with a Windows machine and had Cygwin in their PATH, and it kept picking up the `sh.exe` even though it didn't have access to certain commands.

The solution there was to set `SHELL=cmd.exe` unconditionally.  I couldn't come up with any cross-platform method of determining whether commands were available without throwing a lot of warnings.

enjoy-digital, 2020-03-04:
@mithro, @xobs: i'm no longer using Windows and haven't encountered the issue. Is there something we should change in LiteX or document for this? Or is is related to litex-buildenv?

xobs, 2020-03-04:
I think that it's enough to put up a document somewhere that says if you have this issue, set `SHELL=cmd.exe`.  The issue mostly appears to happen when you have half a Posix environment, but not if it's full-Windows or full-Posix.

enjoy-digital, 2020-03-04:
Thanks @xobs, i just updated the installation instructions (https://github.com/enjoy-digital/litex/commit/d2accbb1a44f82dcf040b52502375cd9c816ea6c) and added a note for that:

> **Note:** On Windows, it's possible you'll have to set `SHELL` environment variable to `SHELL=cmd.exe`.

The wiki also have a copy of this: https://github.com/enjoy-digital/litex/wiki/Installation.

closed 2020-03-04 by enjoy-digital