enjoy-digital/litex #288: trellisboard build on Windows PC

AnttiLukats, 2019-10-25:
2141003	WARNING - The source file 'Python37libsite-packageslitex-0.2.dev0-py3.7.egglitexsoccorescpuexriscverilogVexRiscv.v' in implementation 'impl' doesn't exist.

=
the / \ are wrong but after manual change still not compiling

AnttiLukats, 2019-10-25:
.ERROR - par: A generic DDR interface has been locked to bank(s) 2,3,6, which are not neighbors; therefore, the interface cannot be covered by an ECLK signal per device architecture.

just posting it, this is wha diamond says, if patched to load the top.tcl

enjoy-digital, 2019-10-25:
For https://github.com/enjoy-digital/litex/issues/238, can you post the TCL before and after the manual modification?

enjoy-digital, 2019-10-28:
@daveshah1: i have not yet been able to test with Diamond on Linux but do you know if it was implementing correctly with both Diamond and Yosys/nextpnr or was it only tested with Yosys/nextpnr?

daveshah1, 2019-10-28:
I've only tested with Yosys and nextpnr. This is probably related to a known issue with the choice of clock input pair, but could probably be worked around in Diamond using an ECLKBRIDGECS. I'll look into it.

enjoy-digital, 2019-11-01:
Thanks @daveshah1 for the suggestion, it indeed works and is fixed by https://github.com/litex-hub/litex-boards/commit/5bd8c4d74f11ca3c38dd698f46d993d2871b17b4.

closed 2019-11-01 by enjoy-digital