enjoy-digital/litex #289: Make CSR Memory paged access behavior better visible to developers

sergachev, 2019-10-28:
I spent some time figuring out why one Memory instance was correctly writable only up to 512 bytes and catching very random bugs appearing on writes above 512 B, until I figured out that there is a 'magic' 512 constant here https://github.com/enjoy-digital/litex/blob/ab8af28213c5d744e6749b76452f5053feadcf58/litex/soc/interconnect/csr_bus.py#L89 and that the generated csr.h contains something like xxx_page_write().

Ideas coming to my mind to make this behavior better visible:
 - define page size along with XXX_MEM_BASE (where does 512 B limit actually come from? can we change it?)
 - update https://github.com/timvideos/litex-buildenv/wiki/LiteX-for-Hardware-Engineers#memory
 - generate accessor functions in csr.h which set page number based on address 
 - produce some warnings if memory size is > 512

I do not mind implementing and submitting some of that, just want to get ideas what could be best.

enjoy-digital, 2019-10-28:
Thanks for reporting, i'll have closer look at it in the next days to see what is the best.

enjoy-digital, 2019-11-01:
@sergachev: the actual limitation of 512 bytes is coming from the fact that the CSR region size is 0x800 (4096) and default csr_data_width is 8, so this means the maximum size is 512 bytes. With https://github.com/enjoy-digital/litex/commit/9c3c43c94ade654a1e386ff6dd131195dd6d63d7 i added a check for it. For now this is limitation is not easy to remove, so i would suggest just having this check. In the future we'll see if it's interesting to remove this limitation. Is it fine for you?

sergachev, 2019-11-01:
@enjoy-digital Thank you for the explanation. 9c3c43c94ade654a1e386ff6dd131195dd6d63d7 prohibits creating memories > 512, which is not ideal - they can be used, just with paged access. I would rather make it a warning.

enjoy-digital, 2019-11-01:
@sergachev: can you create a pull request with the behaviour you think should be the best? I'm not using csr.SRAM heavily and would probably recommend using `wishbone.SRAM` for large memories.

sergachev, 2019-11-01:
I do not use csr.SRAM directly, rather wishbone.SRAM indeed, following this guide: [Wishbone ro, logic wo](https://github.com/timvideos/litex-buildenv/wiki/LiteX-for-Hardware-Engineers#wishbone-ro-logic-wo)

```
class RxMem(Module, AutoCSR):
    ...
    def __init__(self):
        ...
        mem = Memory(data_width, depth)
        mem_port = self.specials.port = mem.get_port(write_capable=True)
        self.submodules.wb_sram_if = wishbone.SRAM(mem, read_only=True)
        self.wb_bus = wishbone.Interface(data_width=data_width)
        wb_dec = wishbone.Decoder(self.wb_bus, [(lambda _: 1, self.wb_sram_if.bus)], register=True)
        self.submodules += wb_dec

class SoC(SoCSDRAM):
    ...
    def __init__(self, platform, **kwargs):
        ...
        rx_mem = self.submodules.rx_mem = RxMem()
        self.add_wb_slave(self.mem_map["rx"], rx_mem.wb_bus, 65536)
        self.add_memory_region("rx", self.mem_map["rx"], 65536, type='io')

```

enjoy-digital, 2019-11-01:
OK thanks, i'll look at the guide to see what it's trying to achieve.

enjoy-digital, 2019-11-11:
@sergachev: sorry but i'm still not what what you are trying to achieve (and then if i would recommen another way to reproduce it). Can you explain what kind of ports you want for the Memory, what is accessing it (CPU, logic?) and provide an example i can directly use to reproduce your issue?

sergachev, 2019-11-11:
Oh, I provided the wrong part of the code as an example, sorry. The use case is actually not RxMem, but TxMem - following the [Wishbone r/w, logic ro](https://github.com/timvideos/litex-buildenv/wiki/LiteX-for-Hardware-Engineers#wishbone-rw-logic-ro). So the CPU can read and write over wishbone, logic can only read. 

`
class TxMem(Module, AutoCSR):
    def __init__(self):
		...
        self.specials.mem = Memory(data_width, depth)
        self.specials.rdport = self.mem.get_port()
		...
`

Again, everything works, one just has to know that for the CPU the access is paged, and I found no obvious way to learn about that.

enjoy-digital, 2019-11-15:
Closed via https://github.com/enjoy-digital/litex/pull/307.

closed 2019-11-15 by enjoy-digital