enjoy-digital/litex #290: Where did soc.get_memory_regions() go?

mithro, 2019-10-30:
Looks like `soc.get_memory_regions()` was removed. What function replaces it?

Currently causing my `make.py` to fail with the following;
```
[WARNING] Deprecated, please update : shadow_base replaced by IO regions.
Traceback (most recent call last):
  File "./make.py", line 164, in <module>
    main()
  File "./make.py", line 144, in main
    if "main_ram" in (m[0] for m in soc.get_memory_regions()):
  File "/home/travis/build/mithro/litex-buildenv/third_party/litex/litex/soc/integration/soc_core.py", line 532, in __getattr__
    return Module.__getattr__(self, name)
  File "/home/travis/build/mithro/litex-buildenv/third_party/migen/migen/fhdl/module.py", line 136, in __getattr__
    raise AttributeError("'"+self.__class__.__name__+"' object has no attribute '"+name+"'")
AttributeError: 'BaseSoC' object has no attribute 'get_memory_regions'
```

Looks like there might other references to this in other repositories too;
```
third_party/litesata/examples/make.py:        memory_regions = soc.get_memory_regions()
third_party/liteeth/examples/make.py:    memory_regions = soc.get_memory_regions()
third_party/litescope/examples/make.py:        memory_regions = soc.get_memory_regions()
make.py:    for name, start, size in soc.get_memory_regions():
make.py:            if "main_ram" in (m[0] for m in soc.get_memory_regions()):
```

mithro, 2019-10-30:
Looks like it was removed in https://github.com/enjoy-digital/litex/commit/8be5824e258b84df240d34636aaa539124b92c65#diff-5b5aa398d41ddd6086fbe51e97931f5c

enjoy-digital, 2019-10-30:
Indeed, the aim was to simplify the code and just be to use `soc.mem_regions` dict.

mithro, 2019-10-30:
Seems I just need to do the following?
```patch
ansell@tansell:~/github/timvideos/litex-buildenv-clean$ git diff
diff --git a/make.py b/make.py
index d4fa379..50777e0 100755
--- a/make.py
+++ b/make.py
@@ -91,9 +91,9 @@ def get_bios(builddir, filetype="flash"):
 
 
 def get_bios_maxsize(args, soc):
-    for name, start, size in soc.get_memory_regions():
+    for name, region in soc.memory_regions.items():
         if name == 'rom':
-            return size
+            return region.length
     # FIXME: Hard coded value?
     return 0x8000
 
@@ -141,7 +141,7 @@ def main():
             # SPI flash lack the block RAM resources to run the default
             # firmware. Check whether to use the stub or default firmware
             # should be refined (perhaps soc attribute?).
-            if "main_ram" in (m[0] for m in soc.get_memory_regions()):
+            if "main_ram" in soc.memory_regions:
                 builder.add_software_package("firmware", "{}/firmware".format(os.getcwd()))
             else:
                 builder.add_software_package("stub", "{}/firmware/stub".format(os.getcwd()))
```

enjoy-digital, 2019-10-30:
If you replace `soc.memory_regions` with `soc.mem_regions` it should be fine yes.

mithro, 2019-10-30:
What is the difference between `memory_regions` and `mem_regions`?

enjoy-digital, 2019-10-31:
There is no `soc.memory_regions`, it's `soc.mem_regions`.  I agree the naming is not consistent with the `add_memory_region` method. We could improve that in the future (maybe rename `add_memory_region` to `add_mem_region')

closed 2019-10-31 by enjoy-digital