enjoy-digital/litex #301: SPI flash write from BIOS?

Frank-Buss, 2019-11-07:
I have a MAX1000 board with an external SPI flash. Is it already possible to program this with LiteX? I think I can program it with the Quartus programmer application and indirect programming with a JIC file, but would be good if I could do it with some LiteX command line program.

Maybe the right place would be to enhance the BIOS. It has already some bitbanging functions for writing to the SPI flash. Something like serialflash, similar to serialboot, might be useful, to flash a program instead of loading it to RAM and starting it. lxterm would need to be updated as well, with an additional argument "serial-flash". Would be the same as "serial-boot", just sending a different command for flashing after loading it to RAM.

enjoy-digital, 2019-11-07:
Hi @Frank-Buss,

as you've seen, we already have functions to access the SPI Flash from the BIOS (was added recently), but don't currently have the tools to reprogram it. Having this capability would be very useful for some use-cases and i think @kbeckmann also wanted to implement that. Maybe you should synchronize together on that? Extending the current tools as you are suggesting seems fine.

I'm happy to help testing/integrating that.

Frank-Buss, 2019-11-07:
Ok, I'll try to add the functionality. For testing I implemented the SPI flash for the MAX1000 board, see [here](https://github.com/FrankBuss/litex-boards/commit/4a60cc9a82825d0a59ebb6d98b95649ef5edd1eb). No idea, if everything is right. Then I added a chip erase command for the SPI flash in LiteX, see [here](https://github.com/FrankBuss/litex/commit/468df3c8578c3f440ead269bfc8becc7cc8927df). This is the base for testing it, and I'll erase it as well before writing to it. Currently there is a strange bug:
```
litex> help
LiteX BIOS, available commands:
mr         - read address space
mw         - write address space
mc         - copy address space
fw         - write to flash
ef         - erase whole flash

crc        - compute CRC32 of a part of the address space
ident      - display identifier

reboot     - reset processor
serialboot - boot via SFL

memtest    - run a memory test
litex> fw
fw <offset> <value> [count]
litex> ef
flash erased
litex> mr 0xa0000000 0x40
Memory dump:
0xa0000000  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
0xa0000010  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
0xa0000020  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
0xa0000030  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
litex> fw 0 0x42 3
litex> mr 0xa0000000 0x40
Memory dump:
0xa0000000  00 00 00 21 00 00 00 21 00 00 00 7f ff ff ff ff  ...!...!........
0xa0000010  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
0xa0000020  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
0xa0000030  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
```
But I'm in contact with Antti, I'm just a programmer, he is the hardware guru, will send a PR when this is solved and when I've implemented the SPI flash over serial port function.

kbeckmann, 2019-11-07:
Note that _fw_ writes 32 bit words. You might want to tweak the dummy cycles when you instantiate your SpiFlash class, looks like your results are shifted with one bit.

The flash erase command looks useful. Looks like a fine and isolated PR right there :).

Frank-Buss, 2019-11-08:
Thanks, looks like it needed a different dummy value, works now!
```
litex> ef
flash erased
litex> mr 0xa0000000 32
Memory dump:
0xa0000000  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
0xa0000010  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
litex> fw 0xa0000010 0x78563412 1
litex> fw 0xa0000018 0x21436587 1
litex> mr 0xa0000000 32
Memory dump:
0xa0000000  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
0xa0000010  12 34 56 78 ff ff ff ff 87 65 43 21 ff ff ff ff  .4Vx.....eC!....
```
I'm new to migen (I just know a bit VHDL and Verilog), what does this do in spi_flash.py? Padding some words when reading? Changing the value by one, shifted it exactly by one bit.
But right, this looks like a nice PR :-) now someone can fully test the flash with just the BIOS, I'll submit it independently of the new serial-flash command.

kbeckmann, 2019-11-08:
Nice that you got it to work.

Dummy cycles are needed when using the Fast Read command (and others), but the number of cycles you have to wait before data comes is different between flash chips. For some reason I couldn't find this specified in the datasheet for your flash, but here it's explained for another flash from Winbond. 
[image]

You can see where this is being used here https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/spi_flash.py#L309

enjoy-digital, 2019-11-08:
@mithro was suggesting here that dummy cycles could be configured by software here: https://github.com/enjoy-digital/litex/issues/215#issuecomment-510693778 This is something i wanted to explore but haven't found the time yet. That would be useful to avoid tweaking manually the bistream until finding the right value.

Frank-Buss, 2019-11-08:
I added the flash write support, see [here](https://github.com/enjoy-digital/litex/pull/305). Not sure about some details, like if it should be possible to specify the flash start address. Some FPGA systems store the FPGA configuration in the external SPI flash as well, starting at 0, so the actual program should be stored at an offset. But for the MAX1000 board it works.

Of course, for Quartus someone could use the programming tools. They support indirect flash programming, by first loading a special configuration in the FPGA, which then allows to flash the external SPI flash over JTAG with the programming application. But there might be some other FPGA environments which doesn't have such tools.

I save the data immediately into the flash without loading the image to SDRAM first, because then it should work with systems without external SDRAM as well. For these systems it would be even impossible to flash the external flash without additional hardware, if their FPGA toolchain doesn't have such a function. Of course, for bigger systems with SDRAM, another alternative might be to just load some program like u-boot to SDRAM, and then use this to flash itself, as a second stage bootloader.

I hope I didn't break the other systems, but the changes should be backward compatible.

enjoy-digital, 2020-05-19:
This is now integrated in the BIOS.

closed 2020-05-19 by enjoy-digital